hbldh/bleak #2: Make binary wheels for Windows

hbldh, 2018-04-27:
### Problem:
The `BleakUWPBridge.dll` library is currently built for `Any CPU` and included in the repo and sdist. 
### Solution proposal:
Break it out and include artifacts from the [hbldh/BleakUWPBridge](https://github.com/hbldh/BleakUWPBridge) project's Appveyor builds.

hbldh, 2019-06-24:
Since CI and PyPI deployment has been moved to Azure Pipelines, building of the now very slim BleakUWPBridge (which has been moved to this repo instead of a separate one) can be done upon each build.

Aleksandrovas, 2019-10-13:
Is this problem that BleakUWPBridge.dll library is built for Any CPU, may cause another problem: program that includes bleak, cannot be executed after compiling? I have such a problem while running .py script everything works fine, but once I try to compile it using pyinstaller and run it I'm getting an error:  
```
File "site-packages\bleak\backends\dotnet\__init__.py", line 19, in <module>
System.IO.FileNotFoundException: Unable to find assembly 'BleakUWPBridge'.
   at Python.Runtime.CLRModule.AddReference(String name)
[9580] Failed to execute script test
```
or this problem related to something else?

hbldh, 2020-09-25:
Will not do this. It works fine as it is.

closed 2020-09-25 by hbldh