hbldh/bleak #3: Implement macOS backend

hbldh, 2018-04-27:
Implement CoreBluetooth backend to use for macOS for true cross platform capabilities.

hbldh, 2019-07-03:
@kevincar Is the replacement of `self` necessary [here](https://github.com/hbldh/bleak/blob/develop/bleak/backends/corebluetooth/CentralManagerDelegate.py#L41)? I get it to work without replacing it (I dislike the feeling of overwriting it...), but I want to know if you have more insight into it than me?

kevincar, 2019-07-03:
PyObjC documentation shows an example of creating an objective-c style python class [here](https://pyobjc.readthedocs.io/en/latest/core/intro.html?highlight=class#two-phase-instantiation). I simply followed the examples on that site to avoid any unexplained behavior that may have arisen. 

However, I never tested the code without replacing `self`. So if the code works without replacement, then it sounds like we might be able to remove that.

hbldh, 2019-07-03:
The discovery timout in `connect` is pretty high: 10 seconds. Have you assigned it this high just to ensure no device is missed or have you been experiencing problems with lower threshold?

hbldh, 2019-07-03:
> PyObjC documentation shows an example of creating an objective-c style python class [here](https://pyobjc.readthedocs.io/en/latest/core/intro.html?highlight=class#two-phase-instantiation).

I will not argue with the official documentation's advice to rebind `self`. It stays as it is.

hbldh, 2019-07-03:
Made some minor edits during my first testing of it on a Mac mini mid 2011 running OS X 10.11.6. It works very well! Will be testing more tomorrow!

kevincar, 2019-07-03:
> The discovery timout in connect is pretty high

If you're referring to [this](https://github.com/hbldh/bleak/blob/5443370c072a6fffad9890a92bd0cb78233c6491/bleak/backends/corebluetooth/client.py#L49) then yes. I set it to 10 seconds to allow timing to find devices. However, it would be nice if the discovery timeout within the `connect` function were a user-defined input provided by the BleakClient rather than a hard-coded value.

kevincar, 2019-07-05:
As an update: I previously was utilizing a supplemental framework to get around some issues I was having with assigning delegate objects. With [these changes](https://github.com/kevincar/bleak/commit/a7fa3674bf24659f5ef9af12d80cffc693fc3d2c#diff-bae7e90c74910843b23d05c810f2d17e) the framework can be removed.

hbldh, 2019-07-25:
Hello, sorry for leaving this hanging for some time. I have been very strapped for time available for spare time coding lately...

The [changes](https://github.com/kevincar/bleak/commit/a7fa3674bf24659f5ef9af12d80cffc693fc3d2c#diff-bae7e90c74910843b23d05c810f2d17e) you propose, does that remove the need for the [`corebleak`](https://github.com/hbldh/bleak/tree/develop/bleak/backends/corebluetooth/corebleak) Xcode project alltogether? The [`corebleak.framework`](https://github.com/hbldh/bleak/tree/develop/bleak/backends/corebluetooth/corebleak.framework) as well? Or are they still needed? 

Does something require any compilation at any stage? I am thinking for building binary wheels on CI system.

kevincar, 2019-07-26:
The Xcode project was there to build the corebleak.framework via the [makefile](https://github.com/hbldh/bleak/blob/develop/bleak/backends/corebluetooth/corebleak/Makefile) in the core bleak folder. 

But to answer your question, yes, with these changes, neither the framework nor the Xcode project are necessary. 

I had created this framework for assigning values to read-only properties of objective-c objects before realizing that PyObjC already provides setter functions for this purpose.

hbldh, 2019-08-01:
Integrated [changes](https://github.com/kevincar/bleak/commit/a7fa3674bf24659f5ef9af12d80cffc693fc3d2c#diff-bae7e90c74910843b23d05c810f2d17e into bleak and will merge to master an deploy tomorrow (I cannot test the new changes on a Mac today, so I want to wait for that).

hbldh, 2019-08-02:
This has now been released in version 0.5.0, which is now available from PyPI. Thank you very much. If you want to, add yourself to the `CONTRIBUTORS.rst` file and make a PR for that!

closed 2019-08-02 by hbldh