hbldh/bleak #10: raise RuntimeError('This event loop is already running')

kennyjudo, 2018-06-27:
* bleak version: 
* Python version: 3.6.5
* Operating System: win10 Spyder 3.6.8

### Description
I try the LE device scan example get the error message in Spyder 3.6.8 as title mentioned, the program can should the nearly LE devices name and ID but it out of control after the output, just hang........ 

please anyone can help? 
```

hbldh, 2018-06-27:
I think this is an [issue](https://stackoverflow.com/questions/50243393/runtimeerror-this-event-loop-is-already-running-debugging-aiohttp-asyncio-a) with Spyder I am afraid. It seems that Spyder uses event loops internally to run the code, and this will never work with Bleak.

Try running the code from `cmd` instead and see if it works, or switch to another IDE, e.g. PyCharm where I know that it works.

kennyjudo, 2018-06-27:
Yes. I use the same code in vscode without runtime error, I may switch to
vscode since it’s support anaconda.

hbldh, 2018-06-27:
Good that it works! Closing issue.

closed 2018-06-27 by hbldh