hbldh/bleak #50: Syntax error on Raspberry Pi

kmpm, 2018-11-08:
* bleak version: 0.2.0
* Python version: 3.5.3
* Operating System: Raspbian Stretch

### Description
Trying to run the discovery example from usage section in the documentation in a virtualenv on a raspberry pi with default python 3.5.3 installed.

### What I Did
Saved the discovery code from the docs to `discover.py`

```bash
python3 -m venv venv
. venv/bin/activate
pip install bleak
 python3 discover.py
Traceback (most recent call last):
  File "discover.py", line 2, in <module>
    from bleak import discover
  File "/home/pi/venv/lib/python3.5/site-packages/bleak/__init__.py", line 47, in <module>
    from bleak.backends.bluezdbus.discovery import discover  # noqa
  File "/home/pi/venv/lib/python3.5/site-packages/bleak/backends/bluezdbus/discovery.py", line 6, in <module>
    from bleak.backends.device import BLEDevice
  File "/home/pi/venv/lib/python3.5/site-packages/bleak/backends/device.py", line 30
    return f"{self.address}: {self.name}"
                                        ^
SyntaxError: invalid syntax
```

hbldh, 2018-11-08:
It is due to the fact the python 3.5 does not have the f-string formatting used in bleak, which is a python >= 3.6 package.

I thought all platforms that had BlueZ version capable of running bleak had 3.6 as default. Check what BlueZ version you have and see if that matches reqs in Readme also.

It might be able to relax the 3.6 limit to 3.5, but I have to check through the code first.

kmpm, 2018-11-08:
Hi,
Latest image for RPI has version 5.43 of bluez installed by default.

hbldh, 2018-11-08:
DId a brief check through the code and changed the only 3.6-specific things I found. Uploaded v0.2.2 to PyPI just now, so try it out and see if it performs better.

kmpm, 2018-11-09:
Discover is working now !, Stort tack från en dalslänning.
```
python3 ./discover.py
C3:B7:01:DD:30:99: C3:B7:01:DD:30:99
```

hbldh, 2018-11-09:
Corrected in version 0.2.2 (c1f6581).

closed 2018-11-09 by hbldh