hbldh/bleak #51: dpkg-query not found on all Linux systems

matthewhague, 2018-11-27:
* bleak version: 0.2.2
* Python version: 3.7.1
* Operating System: Arch Linux

### Description

The __init__ function fails with "no such file or directory 'dpkg-query'" when using the module.  Dpkg is not used on Arch Linux, so i would not expect this check to pass.

### What I Did

Started the Python interpreter and attempted
```import bleak```

hbldh, 2018-11-28:
Ah, of course. This was an oversight on my part; thank you for bringing it to my attention!

How would you propose to do this on Arch Linux? To use a call to `pacman`? 
Or try to use `bluetoothctl --version` directly and try to parse that directly since the package is probably already installed? I do not remember why I chose not to do this in the first place, but I think I had a reason.

matthewhague, 2018-11-28:
Cool, thanks! -- i'd already cheated a bit by commenting out the check.  Otherwise it does exactly what i needed, with way less code that the other library i tried.

I had a look around to for a way to check the version in Python, but didn't find anything.  With ctype.util.find_library("bluetooth") you can find the bluetooth library version, but i'm not sure that's what you want.  If you're using bluez via dbus, i'd hope there is a way of querying the availability and version of services, but  don't know enough about dbus to make progress in the time i have.

Using bluetoothctl is probably a reasonable solution.  I'd add some kind of catch for when it's not available.  Perhaps falling back on dpkg-query.  If neither of those work, perhaps just a warning will do?

matthewhague, 2018-11-28:
Looking at the names of the Bluez thing you use, it looks like Bluez [versions](http://0pointer.de/blog/projects/versioning-dbus.html) its interfaces.  Perhaps you don't need to test for a specific version, and can just try opening the interfaces and taking a failure to mean it's not installed.

closed 2018-11-30 by hbldh

hbldh, 2018-11-30:
I will use `bluetoothctl --version` for the time being, and try to do the checking via DBUS in the future.
Version 0.2.3 was incorrect, but it is solved in 0.2.4.

commit 24099809 (2018-11-30): Fixes #52.
Fixes #51.
Fix of string method error in BLEDevice.