hbldh/bleak #52: Timing issue getting characteristics

matthewhague, 2018-11-29:
* bleak version:
* Python version:
* Operating System:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

hbldh, 2018-11-30:
Hm, strange. I will look into this tonight.

hbldh, 2018-11-30:
I *think* the modification I did in version 0.2.4 will solve the problem. I could not reproduce it by your scripts in Ubuntu 18.04 installation, but I think I had misplaced the checking of the property "ServicesResolved" in getting services so this might solve your problem. Try it out and report back here.

closed 2018-11-30 by hbldh

matthewhague, 2018-12-02:
Seems to work, thanks!

commit 24099809 (2018-11-30): Fixes #52.
Fixes #51.
Fix of string method error in BLEDevice.
commit ad3b8860 (2018-11-30): Version 0.2.4

Fixes #52 and #51.