hbldh/bleak #53: from windows.Devices import Enumeration(/n) ModuleNotFoundError: No module named 'windows'

Decee1, 2019-03-15:
Hello
I am facing a issue with using Bleak. I had problems before this issue with "Failed building wheel for pythonnet" but solved it by running "conda install git" 
Now i am facing a problem after installing Bleak where i can't include Windows.Devices
Is this a library that has to be installed? I am running Anaconda if that's a useful infomation

Decee1, 2019-03-15:
https://github.com/computational-metabolomics/dimspy/issues/38
This is what i did to solve "Failed building wheel for pythonnet"

hbldh, 2019-03-18:
I guess you run Python 3.7 then? `pythonnet` does not have pre-built wheels for that version and I know this has been an issue previously.

Regarding the Windows.Devices problem, what Windows version do you run (`python -c "import platform;print(platform.win32_ver())"`)?

Decee1, 2019-03-18:
Well i'll try to downgrade to 3.6 to see if that would solve my problem! 
I tried on my work machine which runs 3.6.2 and it seems like it does run on that machine so that might be the problem! I will give you a heads up later today to see if that is the problem!

bojanpotocnik, 2019-03-18:
I also got this error using Python 3.7.1, however I successfully installed the `pythonnet` with pre-built wheel [pythonnet-2.4.0.dev0-cp37-cp37m-win_amd64.whl](https://ci.appveyor.com/project/pythonnet/pythonnet/branch/master/job/cfqvo5iue7r9uopo/artifacts) (as described on the [pythonnet Wiki - Installation](https://github.com/pythonnet/pythonnet/wiki/Installation#windows-binaries-from-master-or-pull-requests)).

I will wait for @Decee1's result.

hbldh, 2019-03-18:
Interesting. I will have to evaluate this further using the wheel you pointed at @bojanpotocnik.

There is a distinct weakness in the Windows solution now that might be dependent on some dll:s available via `pythonnet` that are only available by accident and not by design right now.

hbldh, 2019-03-18:
In your installed `bleak`-code, modify file `bleak/backends/dotnet/discovery.py` and add the following import on line 15 (before the other clr-imports):
```python
from BleakBridge import Bridge
```
This has solved it for me on Python 3.7.1 with the wheel described above. See if solves the situation for you as well, and if it does I will draft a new release shortly.

bojanpotocnik, 2019-03-18:
@hbldh that solves the problem. Thank you.

closed 2019-03-18 by hbldh

hbldh, 2019-03-18:
New release `0.3.0` has been deployed to PyPI, with the fix detailed above. Try it out and we will reopen the issue if it does not solve your problems.

Decee1, 2019-03-18:
Looks veri positive with the "from BleakBridge import Bridge" I can now connect to my Bluetooth device. Now i am just going to figure out how to use it hehe. It tells me that i cannot find my my Characteristic
Something you can help me with?

Decee1, 2019-03-18:
Trying the Scan example that you provided @hbldh on your project page i successfully scanned 4 Bluetooth devices so i can confirm that adding that line solves the problem
But i would still appreciate if you could help me, idk if i should contact you in another way or just write it in this thread

commit dc1fa1ab (2019-03-18): Version 0.3.0

Fixes #53, a problem for Windows and Python 3.7.