hbldh/bleak #54: wrap_Task

Jeins98, 2019-03-19:
* bleak version: 0.3
* Python version: 3.6  
* Operating System: Windows

### Description

I want to start a notification. But every time I want to start it, I get an error.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

Traceback (most recent call last):
  File "C:/Arbeit/Server_Bleak/temperaturesystem-master_KOPIE/TMRTFlib.py", line 572, in <module>
    loop.run_until_complete(connect_device_bleak(address, loop))
  File "C:\Users\Jonas\AppData\Local\Programs\Python\Python36\lib\asyncio\base_events.py", line 473, in run_until_complete
    return future.result()
  File "C:/Arbeit/Server_Bleak/temperaturesystem-master_KOPIE/TMRTFlib.py", line 565, in connect_device_bleak
    await client.start_notify("23EF6F02-56AC-4519-AE11-0684C77CA30E", callback)
  File "C:\Arbeit\Server_Bleak\temperaturesystem-master_KOPIE\interpreter\lib\site-packages\bleak\backends\dotnet\client.py", line 255, in start_notify
    self._bridge.StartNotify(characteristic, dotnet_callback), loop=self.loop
  File "C:\Arbeit\Server_Bleak\temperaturesystem-master_KOPIE\interpreter\lib\site-packages\bleak\backends\dotnet\utils.py", line 39, in wrap_Task
    raise BleakDotNetTaskError(task.Exception.ToString())
bleak.exc.BleakDotNetTaskError: System.AggregateException: Mindestens ein Fehler ist aufgetreten. ---> System.NullReferenceException: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
   bei BleakBridge.Bridge.<StartNotify>d__8.MoveNext()
   --- Ende der internen Ausnahmestapelüberwachung ---
---> (Interne Ausnahme #0) System.NullReferenceException: Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
   bei BleakBridge.Bridge.<StartNotify>d__8.MoveNext()<---

closed 2019-03-19 by Jeins98

hbldh, 2019-03-19:
Did this resolve itself? Or was it something actually wrong?

Jeins98, 2019-03-19:
Yeah I resolve itself :+1: