hbldh/bleak #55: signature "" on interface ... doesnt exist

Jeins98, 2019-03-19:
* bleak version: 0.3
* Python version: 3.6
* Operating System: Ubuntu

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
Traceback (most recent call last):
  File "/home/jonas/Schreibtisch/TMRTFlib.py", line 572, in <module>
    loop.run_until_complete(connect_device_bleak(address, loop))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 473, in run_until_complete
    return future.result()
  File "/home/jonas/Schreibtisch/TMRTFlib.py", line 552, in connect_device_bleak
    async with BleakClient(address, loop=loop) as client:
  File "/home/jonas/.local/lib/python3.6/site-packages/bleak/backends/client.py", line 35, in __aenter__
    await self.connect()
  File "/home/jonas/.local/lib/python3.6/site-packages/bleak/backends/bluezdbus/client.py", line 76, in connect
    raise BleakError(str(e))
bleak.exc.BleakError: org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

```

hbldh, 2019-03-22:
I also got this error when I tried at first. Then I performed a discovery operation first (see [discover example](https://github.com/hbldh/bleak/blob/master/examples/discover.py)), and after that it works. It's like the DBus interface must be "started" and have stored a entry about the peripheral you want to connect to.

Try this and it it works I will add a discover action when `connect` is called.

Jeins98, 2019-04-03:
Thanks for awnser. I will try it these days and will give you a statemant

Jeins98, 2019-04-03:
40:B8:9A:CE:6A:52: dev_40_B8_9A_CE_6A_52
A4:C1:38:14:43:32: dev_A4_C1_38_14_43_32
40:16:3B:DB:96:E3: dev_40_16_3B_DB_96_E3
6F:63:0A:C8:7C:14: dev_6F_63_0A_C8_7C_14
00:1A:22:0B:43:E9: dev_00_1A_22_0B_43_E9
4B:63:5E:85:8B:D2: dev_4B_63_5E_85_8B_D2
51:B9:7C:1C:84:90: dev_51_B9_7C_1C_84_90
Traceback (most recent call last):
  File "/home/jonas/.local/lib/python3.6/site-packages/bleak/backends/bluezdbus/client.py", line 74, in connect
    ).asFuture(self.loop)
txdbus.error.RemoteError: org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jonas/Schreibtisch/TMRTFlib.py", line 575, in <module>
    loop.run_until_complete(connect_device_bleak(address, loop))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 473, in run_until_complete
    return future.result()
  File "/home/jonas/Schreibtisch/TMRTFlib.py", line 552, in connect_device_bleak
    async with BleakClient(address, loop=loop) as client:
  File "/home/jonas/.local/lib/python3.6/site-packages/bleak/backends/client.py", line 35, in __aenter__
    await self.connect()
  File "/home/jonas/.local/lib/python3.6/site-packages/bleak/backends/bluezdbus/client.py", line 76, in connect
    raise BleakError(str(e))
bleak.exc.BleakError: org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

I also get this error

hbldh, 2019-04-03:
Could you provide a code snippet showing how you create the client and call connect? Try to extract a minimal example that reproduces the error, Also, what version of Ubuntu do you use and what version of BlueZ?

Jeins98, 2019-04-09:
Sorry for late awnser. Here is my Code Snippet:

   async def measure_battery(self, loop):
        async with BleakClient(self.__address, loop=loop) as client:

            await client.start_notify(self.__BtriggerCharacteristicUUID, self.callback_battery)
            await asyncio.sleep(2.0, loop=loop)
            await client.stop_notify(self.__BtriggerCharacteristicUUID)

async def scan_bleak():
    devices = await discover()
    for d in devices:
        print(d)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(scan_bleak())
    s = Sensor("EC:67:FD:C2:79:1B", 5)
    loop.run_until_complete(s.measure_temperature(loop))

hbldh, 2019-04-09:
Ah, this is a bit of a problematic case...

In BlueZ 5.48 they removed the Battery Service (0000180f-0000-1000-8000-00805f9b34fb)  from the has been moved to interface `org.bluez.Battery1` instead of as a regular service. This has been remedied in Bleak 0.4.0 (which isn't released yet, still in develop branch), at least for the read case; however I have not fixed notifications from the battery service... I will look into this and get back to you.

hbldh, 2019-04-09:
After I release version 0.4.0 and you have installed that, try this instead:
```
async def measure_battery(self, loop):
     async with BleakClient(self.__address, loop=loop) as client:
         battery_level = await client.read_gatt_char(self.__BtriggerCharacteristicUUID)
         print("Battery Level: {0}%".format(int(battery_level[0])))
```
Notifications on battery level will not work in BlueZ right now; I do not know how to fix that right now. You will have to settle for a regular read.

closed 2019-04-09 by hbldh

Jeins98, 2019-04-24:
Hi sorry, but I don't use the Battery UUID. It's a Custom Service and it don't work.

karlp, 2022-12-01:
I'm getting this still with bluez 5.64 on openwrt (dbus, but not systemd) using the script: https://github.com/etactica/host_utilities/blob/feature/bluetooth_ota_python_bleak/bluetooth_ota_apploader_python_bleak/sl-ota-apploader.py

It happily scans and finds the device, then .... fails to connect.    This works fine on fedora desktop with bluez5.65, and it worked "back then" before I updated fedora as well.

Using bluetoothctl, I can see that I must _leave scan on_ to get a connection to succeed?  if I do "scan le" my device appears, but if I do "scan off" (to stop the console spam) and then "connect blah" it will fail.  if I leave scan on, I can connect just fine

commit 525d895f (2019-03-23): Bugfixes and finished BlueZ update.

Fixed Battery level bug for BlueZ >=5.48
Also fixed #55 by adding a discover call before connecting.
commit 2a243779 (2019-03-24): OOP representation of GATT objects.

Updated DBus backend to use new object structure
Updated .NET backend to use new object structure
Fixed Battery level bug for BlueZ >=5.48
Also fixed #55 by adding a discover call before connecting.
Fix for .NET stop_notify
Slimmed down .NET bridge considerably thanks to .NET code moved from Bridge into bleak instead.
Added BleakUWPBridge project to this repo again.
New example: service_explorer.py