hbldh/bleak #57: Windows: use BluetoothLEAdvertisementWatcher instead of generic DeviceWatcher

dlech, 2019-04-01:
Since there seem to be major changes currently being worked on, I will bring this up in an issue instead of making a pull request.

---

* bleak version: https://github.com/virantha/bleak
* Python version: 3.6
* Operating System: Windows 10

### Description

Trying to get manufacturer data from BLE advertising packets. This does not seem to be possible using `Windows.Devices.Enumeration.DeviceWatcher`.

### What I Did

Replaced `Windows.Devices.Enumeration.DeviceWatcher` with `Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisementWatcher` in https://github.com/virantha/bleak/commit/c6815586e5964396d22906a94eb5af2482830d25.

hbldh, 2019-04-09:
I will implement this improvement later on. 
In version 0.4.0 I have moved most of the code from the BleakBridge project into this and ported most of the C# code to pythonnet code instead. I will migrate the discovery code for version 0.4.1 or similar.

hbldh, 2019-04-12:
@dlech Integrated this into bleak in commit 95e66f853e60851a1a04a74503f1515282d69ef9. Will try it out for a while before I merge it into master. I have had some problems with it, missing detection of devices some times. Something you have seen?

dlech, 2019-04-13:
I don't recall seeing such problems. I don't use Windows very much though. I can try testing it more.

commit e14b5bc9 (2019-05-17): Version 0.4.1

Merged #61: improvements and fixes for multiple issues for BlueZ backend
Fixes #57
Fixes #59
Additional documentation fixes
closed 2019-05-17 by hbldh