hbldh/bleak #58: Can't get all

skjerns, 2019-04-09:
* bleak version: 0.3.0
* Python version: 3.7
* Operating System: Win 10

### Description

I am trying to communicate with a non-standard service. I have the UUID, which I can also read from `client.get_services`, however, I cannot seem to get a handle of the corresponding characteristic. I do know the characteristic UUID from using Bluetooth LE Explorer.

However, when calling `client.characteristics`, this UUID does not appear. If I try to obtain them via the service handle via `service.GetAllCharacteristics()`, I get a `FileLoadException`. If I try to directly access the characteristic via `client.write_gatt_char`, I get a `UUID not found`

```Python
mac ='d1:10:31:fa:3a:22'
loop = asyncio.get_event_loop()
client = BleakClient(address, loop=loop)
await client.connect()

rw_service = '49535343-fe7d-4ae5-8fa9-9fafd205e455'
rw_charac = '49535343-1e4d-4bd9-ba61-23c647249616'

# let's see if we can find the service:
await client.get_services()
assert rw_service in client.services # True

# first I try to access the characteristic directly
client.write_gatt_char(rw_charac, b'~HELLO', response=True)
# BleakError: Characteristic 49535343-1e4d-4bd9-ba61-23c647249616 was not found!

# second I try to enumerate all characteristics to get a handle on the char
# ..... many UUIDS, but none of the above UUID are in .characteristics
assert rw_charac in client.characteristics # failed

# third I try to enumerate characteristics of the rw_service
service = client.services[rw_service]
characts = service.GetAllCharacteristics()
#FileLoadException: The process cannot access the file because it is being used by another process. (Exception from HRESULT: 0x80070020)
#   at Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService.GetAllCharacteristics()
```

Am I doing anything wrong? Do you have a suggestion how I can send simple command to my custom r/w service?

hbldh, 2019-04-10:
You have forgotten an `await` before `client.write_gatt_char`; that method is still a coroutine. It might be the reason for the FileLoadException, but I am not sure.

I just released verion 0.4.0 of bleak; install that from PyPI and run the [service_explorer.py](https://github.com/hbldh/bleak/blob/master/examples/service_explorer.py) example on your peripheral and show me your output here. It should display all services and characteristics discovered on the peripheral, similar to Bluetooth LE Explorer.

In 0.4.0, the `client.services` now returns a (BleakGATTServiceCollection)[https://bleak.readthedocs.io/en/latest/api.html#bleak.backends.service.BleakGATTServiceCollection] which should be simpler to handle than the .NET objects returned in v0.4.0.

skjerns, 2019-04-10:
Thanks for the fast reply!

With this output, all services and their characteristics get printed as expected :) and it doesn't throw the `UUID not found` Exception anymore when paging the rw_charac.

However, when I request a write operation with response, I get no response:

```
await client.write_gatt_char(rw_charac, b'~HELLO', response=True)
# nothing happens.
```

hbldh, 2019-04-10:
The `response=True` mean that the `write` method is used rather than the `write-without-response`. The response does not contain anything special though (https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattwriteresult), so nothing is returned by the `write_gatt_char` method.

You can check the value of your characteristic by a read afterwards to verify that your data has been written:
```python
value = await client.read_gatt_char(rw_charac)
print("Pre-Write Value: {0}".format(value))
await client.write_gatt_char(rw_charac, b'~HELLO', response=True)
value = await client.read_gatt_char(rw_charac)
print("Post-Write Value: {0}".format(value))
```

skjerns, 2019-04-10:
Somehow I can't read from this characteristic.

`BleakError: Could not read characteristic value for 49535343-1e4d-4bd9-ba61-23c647249616: 2`

hbldh, 2019-04-11:
Ah, i did not see that; of course it fails. The read was not important anyway, it was just meant to visualize what the write achieved.

The issue in general is to be regarded as closed then? There is nothing left to solve of the original request is there?

skjerns, 2019-04-11:
Yes, I'll open another issue for the non-response :) thank's!

closed 2019-04-11 by skjerns