hbldh/bleak #60: GattCommunicationStatus.Unreachable in BleakClientDotNet.get_services()

dlech, 2019-04-13:
* bleak version: b658f528f2e6a7f0dc6857d8c109029112a3bba6
* Python version: 3.6.8
* Operating System: Windows 10 1809

### Description

Testing out the develop branch. `BleakClientDotNet.connect()` throws `BleakDotNetTaskError` because `GetGattServicesAsync()` in `BleakClientDotNet.get_services()` returns `GattCommunicationStatus.Unreachable`. 

This seems to be a timing issue. For some reason, `GetGattServicesAsync` returns before it actually connects to the device. If I add a delay before calling `GetGattServicesAsync`, it works more reliably. It also seems to work if I add a retry instead of a delay.

```python

            async def GetGattServicesAsync():
                return await wrap_IAsyncOperation(
                    IAsyncOperation[GattDeviceServicesResult](
                        self._requester.GetGattServicesAsync()
                    ),
                    return_type=GattDeviceServicesResult,
                    loop=self.loop,
                )
            
            services_result = await GetGattServicesAsync()
            
            if services_result == GattCommunicationStatus.Unreachable:
                # one retry because Windows is goofy
                services_result = await GetGattServicesAsync()

            if services_result.Status != GattCommunicationStatus.Success:
                raise BleakDotNetTaskError("Could not get GATT services.")
```

[This thread](https://social.msdn.microsoft.com/Forums/en-US/1810b32a-ea5f-4183-8f3f-d4a5b76401f9/uwp-ble-gatt-getgattservicesasync-method-returns-gattcommunicationstatusunreachable?forum=wpdevelop) lead me to [this thread](https://social.msdn.microsoft.com/Forums/vstudio/en-US/8a452e2c-26a1-4911-8f95-2b3c5c9f605b/getcharacteristicsasync-sometimes-fails-with-hresult-0x80070020?forum=csharpgeneral) which got me wondering if this behavior has something to do with needing a single threaded apartment on Windows.

dlech, 2019-04-14:
After some more testing, the single retry method doesn't always work. It would probably be better with multiple retries and a timeout.

hbldh, 2019-04-18:
Might this be an effect of the `BluetoothLEAdvertisementWatcher` discovery change? Or maybe even the `FromBluetoothAddressAsync` change? Have you experienced the same problems prior to this change?

I did notice that during the discovery phase and reading of manufacturer data, my test device stopped blinking (which it does to signal that it is possible to connect to). Does Windows create a connection during the reading of the manufacturer data which interferes with the services detecting later on?

zyv, 2019-09-15:
I think one of my library users experiences the same problem - he reported that if he just comments out the exception (I'm not using discovered services anyways) - everything works.

hbldh, 2020-06-02:
I think this is solved in later versions. Will close for now.

closed 2020-06-02 by hbldh