hbldh/bleak #62: Error installing bleak on Windows 10

gmussi, 2019-04-29:
* bleak version: 0.4.0
* Python version: Python 3.7.0 (v3.7.0:1bf9cc5093, Jun 27 2018, 04:59:51) [MSC v.1914 64 bit (AMD64)] on win32
* Operating System: Windows 10

### Description

When installing bleak, there was an error importing basetsd.h. After installing Windows SDK, the installation output changed, but the error persists.

### What I Did
When I run `pip install bleak` the following stack trace appears:

```
PS C:\Users\GuilhermeMussi\Desktop\esp32py> pip install bleak
Collecting bleak
  Using cached https://files.pythonhosted.org/packages/3a/0e/ce9655465ac5455f9bbec5cb41b007a26bd889792fd1d4963f6f99d88b1a/bleak-0.4.0-py2.py3-none-any.whl
Collecting pythonnet; platform_system == "Windows" (from bleak)
  Using cached https://files.pythonhosted.org/packages/89/3b/a22cd45b591d6cf490ee8b24d52b9db1f30b4b478b64a9b231c53474731e/pythonnet-2.3.0.tar.gz
Installing collected packages: pythonnet, bleak
  Running setup.py install for pythonnet ... error
    ERROR: Complete output from command 'c:\users\guilhermemussi\appdata\local\programs\python\python37\python.exe' -u -c 'import setuptools, tokenize;__file__='"'"'C:\\Users\\GUILHE~1\\AppData\\Local\\Temp\\pip-install-pbsf463r\\pythonnet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\GUILHE~1\AppData\Local\Temp\pip-record-v4n1x0hi\install-record.txt' --single-version-externally-managed --compile:
    ERROR: running install
    running build
    running build_ext
    Checking for updates from https://www.nuget.org/api/v2/.
    Currently running NuGet.exe 3.5.0.
    Updating NuGet.exe to 4.9.4.
    Update successful.
    MSBuild auto-detection: using msbuild version '16.0.461.62831' from 'C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Current\bin'.
    Restoring NuGet package NUnit.3.6.0.
    Restoring NuGet package NUnit.ConsoleRunner.3.6.0.
    Restoring NuGet package UnmanagedExports.1.2.7.
    Adding package 'NUnit.ConsoleRunner.3.6.0' to folder 'C:\Users\GuilhermeMussi\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\packages'
    Adding package 'UnmanagedExports.1.2.7' to folder 'C:\Users\GuilhermeMussi\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\packages'
    Adding package 'NUnit.3.6.0' to folder 'C:\Users\GuilhermeMussi\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\packages'
    Added package 'UnmanagedExports.1.2.7' to folder 'C:\Users\GuilhermeMussi\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\packages'
    Added package 'NUnit.ConsoleRunner.3.6.0' to folder 'C:\Users\GuilhermeMussi\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\packages'
    Added package 'NUnit.3.6.0' to folder 'C:\Users\GuilhermeMussi\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\packages'

    NuGet Config files used:
        C:\Users\GuilhermeMussi\AppData\Roaming\NuGet\NuGet.Config

    Feeds used:
        C:\Users\GuilhermeMussi\.nuget\packages\
        https://api.nuget.org/v3/index.json

    Installed:
        3 package(s) to packages.config projects
    clang.exe: warning: c:\users\guilhermemussi\appdata\local\programs\python\python37\Include: 'linker' input unused
    In file included from c:\users\guilhermemussi\appdata\local\programs\python\python37\Include\Python.h:8:
    c:\users\guilhermemussi\appdata\local\programs\python\python37\Include/pyconfig.h:215:10: fatal error: 'basetsd.h' file not found
    #include <basetsd.h>
             ^
    1 error generated.
    Traceback (most recent call last):
      File "tools\geninterop\geninterop.py", line 292, in <module>
        sys.exit(main())
      File "tools\geninterop\geninterop.py", line 272, in main
        python_h = preprocess_python_headers()
      File "tools\geninterop\geninterop.py", line 192, in preprocess_python_headers
        for line in _check_output(cmd).splitlines():
      File "tools\geninterop\geninterop.py", line 41, in _check_output
        output = subprocess.check_output(*args, **kwargs)
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\subprocess.py", line 376, in check_output
        **kwargs).stdout
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\subprocess.py", line 468, in run
        output=stdout, stderr=stderr)
    subprocess.CalledProcessError: Command '['clang', '-I', 'tools\\geninterop\\fake_libc_include', 'c:\\users\\guilhermemussi\\appdata\\local\\programs\\python\\python37\\Include', '-D', '__attribute__(x)=', '-D', '__inline__=inline', '-D', '__asm__=;#pragma asm', '-D', '__int64=long long', '-E', 'c:\\users\\guilhermemussi\\appdata\\local\\programs\\python\\python37\\Include\\Python.h']' returned non-zero exit status 1.
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\GUILHE~1\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\setup.py", line 405, in <module>
        zip_safe=False,
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\site-packages\setuptools\__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\core.py", line 148, in setup
        dist.run_commands()
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\site-packages\setuptools\command\install.py", line 61, in run
        return orig.install.run(self)
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\command\install.py", line 545, in run
        self.run_command('build')
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\command\build.py", line 135, in run
        self.run_command(cmd_name)
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\command\build_ext.py", line 339, in run
        self.build_extensions()
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\command\build_ext.py", line 448, in build_extensions
        self._build_extensions_serial()
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\distutils\command\build_ext.py", line 473, in _build_extensions_serial
        self.build_extension(ext)
      File "C:\Users\GUILHE~1\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\setup.py", line 191, in build_extension
        subprocess.check_call([sys.executable, geninterop, interop_file])
      File "c:\users\guilhermemussi\appdata\local\programs\python\python37\lib\subprocess.py", line 328, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['c:\\users\\guilhermemussi\\appdata\\local\\programs\\python\\python37\\python.exe', 'tools\\geninterop\\geninterop.py', 'src\\runtime\\interop37.cs']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command "'c:\users\guilhermemussi\appdata\local\programs\python\python37\python.exe' -u -c 'import setuptools, tokenize;__file__='"'"'C:\\Users\\GUILHE~1\\AppData\\Local\\Temp\\pip-install-pbsf463r\\pythonnet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\GUILHE~1\AppData\Local\Temp\pip-record-v4n1x0hi\install-record.txt' --single-version-externally-managed --compile" failed with error code 1 in C:\Users\GUILHE~1\AppData\Local\Temp\pip-install-pbsf463r\pythonnet\
```

I am not sure what else I can do. Any help would be appreciated.

dlech, 2019-04-29:
I don't think pythonnet officially supports Python 3.7 yet. If you use Python 3.6, it should downloaded a binary wheel instead of trying to build pythonnet from source.

hbldh, 2019-04-29:
As @dlech says, there are no binary wheels for Python 3.7. Building from source requires a non-trivial setup of Visual Studio/msbuild and something I recommend only if you have some time over...
Otherwise, check out #53 for details about how to install a non-PyPI wheel.

amitlalaawaz, 2019-05-13:
Hello,

Please help to fix this error -

(base) D:\download\TI_IOT\BLEAK>python sensortag.py
poll 2000.000 ms took 2000.000 ms: timeout
poll took 156.000 ms: 1 events
poll took 219.000 ms: 1 events
poll took 16.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 15.000 ms: 1 events
poll took 16.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 15.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 16.000 ms: 1 events
Traceback (most recent call last):
  File "sensortag.py", line 153, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "C:\Users\amit\Anaconda3\lib\asyncio\base_events.py", line 473, in run_until_complete
    return future.result()
  File "sensortag.py", line 100, in run
    async with BleakClient(address, loop=loop) as client:
  File "C:\Users\amit\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Users\amit\Anaconda3\lib\site-packages\bleak\backends\client.py", line 41, in __aenter__
    await self.connect()
  File "C:\Users\amit\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Users\amit\Anaconda3\lib\site-packages\bleak\backends\dotnet\client.py", line 115, in connect
    await self.get_services()
  File "C:\Users\amit\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Users\amit\Anaconda3\lib\site-packages\bleak\backends\dotnet\client.py", line 202, in get_services
    "Could not get GATT characteristics for {0}.".format(service)
bleak.exc.BleakDotNetTaskError: Could not get GATT characteristics for Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService.

Thanks!

hbldh, 2019-05-17:
@gmussi Did you get this to work? May 4, the version 2.4 of pythonnet was released, and it has Python 3.7 wheels so try to install that instead.

gmussi, 2019-05-23:
@hbldh i follows the sugestion of @dlech and it worked fine.

closed 2019-05-23 by gmussi