hbldh/bleak #63: Could not get GATT characteristics for Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService.

amitlalaawaz, 2019-05-13:
* bleak version: latest
* Python version: 3.6.7
* Operating System: Win 10 above 17xxx

### Description
Trying to run example python code.

(base) C:\Users\amit\Documents\GitHub\bleak\examples>python sensortag.py
poll 2000.000 ms took 2000.000 ms: timeout
poll took 172.000 ms: 1 events
poll took 172.000 ms: 1 events
poll took 16.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 16.000 ms: 1 events
poll took 31.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 15.000 ms: 1 events
poll took 16.000 ms: 1 events
poll took 0.000 ms: 1 events
poll took 16.000 ms: 1 events
Traceback (most recent call last):
  File "sensortag.py", line 156, in <module>
    loop.run_until_complete(run(address, loop, True))
  File "C:\Users\amit\Anaconda3\lib\asyncio\base_events.py", line 473, in run_until_complete
    return future.result()
  File "sensortag.py", line 103, in run
    async with BleakClient(address, loop=loop) as client:
  File "C:\Users\amit\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Users\amit\Anaconda3\lib\site-packages\bleak\backends\client.py", line 41, in __aenter__
    await self.connect()
  File "C:\Users\amit\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Users\amit\Anaconda3\lib\site-packages\bleak\backends\dotnet\client.py", line 115, in connect
    await self.get_services()
  File "C:\Users\amit\Anaconda3\lib\asyncio\coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "C:\Users\amit\Anaconda3\lib\site-packages\bleak\backends\dotnet\client.py", line 202, in get_services
    "Could not get GATT characteristics for {0}.".format(service)
bleak.exc.BleakDotNetTaskError: Could not get GATT characteristics for Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService.

### What I Did
Running

C:\Users\amit\Documents\GitHub\bleak\examples>python sensortag.py
```
Paste the command(s) you ran and the output.
  await self.get_services()
  File "C:\Users\amit\Anaconda3\lib\site-packages\bleak\backends\dotnet\client.py", line 202, in get_services
    "Could not get GATT characteristics for {0}.".format(service)
bleak.exc.BleakDotNetTaskError: Could not get GATT characteristics for Windows.Devices.Bluetooth.GenericAttributeProfile.GattDeviceService.
![1111](https://user-images.githubusercontent.com/37605691/57596682-923ee500-7511-11e9-83de-10982bb737f2.JPG)

hbldh, 2019-05-17:
Do you own a CC2650 sensor tag to connect to? DId you enter its address in the `sensortag.py` script?

amitlalaawaz, 2019-05-17:
Hello !! Its connecting now. It works only one Bluetooth connection at a time. Code is perfect!

Thanks!
Amit Lal

amitlalaawaz, 2019-05-20:
Closing this issue now. THank you

closed 2019-05-20 by amitlalaawaz

Piruzzolo, 2021-05-10:
> Hello !! Its connecting now. It works only one Bluetooth connection at a time. Code is perfect!
> 
> Thanks!
> Amit Lal

What did you do to solve the issue?