hbldh/bleak #64: Unable to Discover/Connect to iOS Devices

stevennyman, 2019-05-15:
* bleak version: 0.4.0
* Python version: 3.7.3
* Operating System: Windows 10 version 1809

### Description

I tried running the `discover.py` example, and my iOS 12.1.4 iPhone SE was not recognized. I had the Bluetooth menu open on the device so it would be discoverable.

I tried getting the device's services by using the `get_services.py` example with the iOS device's Bluetooth address filled into the script. The device was not found.
Traceback for `get_services.py`:

```python
Traceback (most recent call last):
  File "C:/Users/xx/services.py", line 21, in <module>
    loop.run_until_complete(print_services(mac_addr, loop))
  File "C:\Users\xx\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "C:/Users/xx/services.py", line 14, in print_services
    async with BleakClient(mac_addr, loop=loop) as client:
  File "C:\Users\xx\AppData\Roaming\Python\Python37\site-packages\bleak\backends\client.py", line 41, in __aenter__
    await self.connect()
  File "C:\Users\xx\AppData\Roaming\Python\Python37\site-packages\bleak\backends\dotnet\client.py", line 97, in connect
    "Device with address {0} was " "not found.".format(self.address)
bleak.exc.BleakError: Device with address f0:xx:xx:xx:xx:xx was not found.
```

stevennyman, 2019-05-15:
Note that when typically attempting to pair Windows and iOS a code and confirmation prompt are displayed on both devices.

hbldh, 2019-05-17:
Bleak is a Bluetooth Low Energy communication package, only capable of communicating to devices exposing services via that technology. Pairing with you iOS device would typically be done via regular Bluetooth, so this package is regrettably not capable of helping you.

stevennyman, 2019-05-22:
The documentation for the API I would like to work with specifically mentions Bluetooth LE: https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleNotificationCenterServiceSpecification/Introduction/Introduction.html. @hbldh

hbldh, 2019-05-25:
Ok, apparently I was permature in closing this. I did not understand what you actually wanted to do.

I can detect multiple iOS devices using the `discover` method, but strangely enough not my own. I was also capable of connecting to an iOS device and request some data, but not from all characteristics, so I think there needs to be some modifications in the code to allow for handling access.

I will try to find out why all devices does not show up.

hbldh, 2020-09-25:
It is possible to connect and get services from iOS devices with Bleak now. This is output from `service_explorer.py` example:

```
[Service] 00001800-0000-1000-8000-00805f9b34fb: Generic Access Profile
	[Characteristic] 00002a00-0000-1000-8000-00805f9b34fb: (Handle: 2) (read) | Name: , Value: b'iPhone' 
	[Characteristic] 00002a01-0000-1000-8000-00805f9b34fb: (Handle: 4) (read) | Name: , Value: b'@\x00' 
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (Handle: 7) (indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 9) | Value: b'\x02\x00' 
[Service] d0611e78-bbb4-4591-a5f8-487910ae4366: Apple Continuity Service
	[Characteristic] 8667556c-9a37-4c91-84ed-54ee27d90049: (Handle: 11) (write,notify,extended-properties,reliable-writes) | Name: , Value: None 
		[Descriptor] 00002900-0000-1000-8000-00805f9b34fb: (Handle: 13) | Value: b'\x01\x00' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 14) | Value: b'\x00\x00' 
[Service] 9fa480e0-4967-4542-9390-d343dc5d04ae: Apple Nearby Service
	[Characteristic] af0badb1-5b99-43cd-917a-a77bc549e3cc: (Handle: 16) (write,notify,extended-properties,reliable-writes) | Name: , Value: None 
		[Descriptor] 00002900-0000-1000-8000-00805f9b34fb: (Handle: 18) | Value: b'\x01\x00' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 19) | Value: b'\x00\x00' 
[Service] 0000180f-0000-1000-8000-00805f9b34fb: Battery Service
	[Characteristic] 00002a19-0000-1000-8000-00805f9b34fb: (Handle: 21) (read,notify) | Name: , Value: b'Could not get GATT characteristics for 00002a19-0000-1000-8000-00805f9b34fb: ProtocolError (Error: 0x05: Authentication Failure)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 23) | Value: b'\x00\x00' 
[Service] 00001805-0000-1000-8000-00805f9b34fb: Current Time Service
	[Characteristic] 00002a2b-0000-1000-8000-00805f9b34fb: (Handle: 25) (read,notify) | Name: , Value: b'Could not get GATT characteristics for 00002a2b-0000-1000-8000-00805f9b34fb: ProtocolError (Error: 0x05: Authentication Failure)' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 27) | Value: b'\x00\x00' 
	[Characteristic] 00002a0f-0000-1000-8000-00805f9b34fb: (Handle: 28) (read) | Name: , Value: b'Could not get GATT characteristics for 00002a0f-0000-1000-8000-00805f9b34fb: ProtocolError (Error: 0x05: Authentication Failure)' 
[Service] 0000180a-0000-1000-8000-00805f9b34fb: Device Information
	[Characteristic] 00002a29-0000-1000-8000-00805f9b34fb: (Handle: 31) (read) | Name: , Value: b'Apple Inc.' 
	[Characteristic] 00002a24-0000-1000-8000-00805f9b34fb: (Handle: 33) (read) | Name: , Value: b'iPhone10,5' 
[Service] 7905f431-b5ce-4e99-a40f-4b1e122d00d0: Apple Notification Center Service
	[Characteristic] 69d1d8f3-45e1-49a8-9821-9bbdfdaad9d9: (Handle: 36) (write,extended-properties,reliable-writes) | Name: , Value: None 
		[Descriptor] 00002900-0000-1000-8000-00805f9b34fb: (Handle: 38) | Value: b'\x01\x00' 
	[Characteristic] 9fbf120d-6301-42d9-8c58-25e699a21dbd: (Handle: 39) (notify) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 41) | Value: b'\x00\x00' 
	[Characteristic] 22eac6e9-24d6-4bb5-be44-b36ace7c7bfb: (Handle: 42) (notify) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 44) | Value: b'\x00\x00' 
[Service] 89d3502b-0f36-433a-8ef4-c502ad55f8dc: Apple Media Service
	[Characteristic] 9b3c81d8-57b1-4a8a-b8df-0e56f7ca51c2: (Handle: 46) (write,notify,extended-properties,reliable-writes) | Name: , Value: None 
		[Descriptor] 00002900-0000-1000-8000-00805f9b34fb: (Handle: 48) | Value: b'\x01\x00' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 49) | Value: b'\x00\x00' 
	[Characteristic] 2f7cabce-808d-411f-9a0c-bb92ba96c102: (Handle: 50) (write,notify,extended-properties,reliable-writes) | Name: , Value: None 
		[Descriptor] 00002900-0000-1000-8000-00805f9b34fb: (Handle: 52) | Value: b'\x01\x00' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 53) | Value: b'\x00\x00' 
	[Characteristic] c6b2f38c-23ab-46d8-a6ab-a3a870bbd5d7: (Handle: 54) (read,write,extended-properties,reliable-writes) | Name: , Value: b'Could not get GATT characteristics for c6b2f38c-23ab-46d8-a6ab-a3a870bbd5d7: ProtocolError (Error: 0x05: Authentication Failure)' 
		[Descriptor] 00002900-0000-1000-8000-00805f9b34fb: (Handle: 56) | Value: b'\x01\x00' 
```

I have not succeded in pairing with one yet though (I think it requires PIN codes and such like, which Bleak does not handle yet), so some characteristics are off limits. However, I will close the issue since it is possible at least.

closed 2020-09-25 by hbldh

athar1001, 2021-02-06:
i  am trying to connect IOS device with service_explorer.py but getting timeout error can you please help on this issue.

hbldh, 2021-02-19:
@athar1001 Try setting the timeout to a higher value. Change line 28 of `service_explorer.py` to
```python
async with BleakClient(address, timeout=30) as client:
```

vaskomichal, 2021-02-22:
> i am trying to connect IOS device with service_explorer.py but getting timeout error can you please help on this issue.

Did you fix the problem? I paired my device with raspberry but still I had the same error Device with address XX:XX:XX:XX:XX:XX was not found. was not found.

PR #269 (2020-09-02): First drafts of "Just Works" pairing for .NET and BlueZ backends
Might fix #64
Might fix #77
Might fix #203 
Fixes #218