hbldh/bleak #65: Google Cloud integration

amitlalaawaz, 2019-05-20:
* bleak version: current 
* Python version: 3.6
* Operating System: Win10

### Description
Trying to modify service_explorer python code and adding GCP connection.

## BELOW FUNCTION USES PROJECT SPECIFIC INFORMATION TO SEND DATA TO PUB/SUB. CREATE YOUR OWN CREDENTIALS OF PROJECT,PUB/SUB TOPIC AND SERVICE ACCOUNT FILE. 
def sendDataToPubSub(data):
    project_id = 'xxxxxxxxxxxxxxxxxxxxxxxxxxxx'
    topic_name = 'cchain_test_topic'
    cred = service_account.Credentials.from_service_account_file(filename = '//us-con-gcp-npr-0xxxxxxxxxxxx.json')
    publisher = pub.PublisherClient(credentials = cred)
    topic_path = publisher.topic_path(project_id, topic_name)
    publisher.publish(topic_path, data="",sensor_timestamp=str(data.get('timestamp')) , latitude = str(data.get('latitude')) ,longitude = str(data.get('longitude')) ,ambient_temperature = str(data.get('ambient_temperature')) ,relative_humidity = str(data.get('relative_humidity')) ,barometeric_pressure = str(data.get('barometeric_pressure')) ,magnetometer_data_x = str(data.get('magnetometer_data_x')) ,magnetometer_data_y = str(data.get('magnetometer_data_y')) ,magnetometer_data_z = str(data.get('magnetometer_data_z')), accerometer_data_x = str(data.get('accerometer_data_x')),accerometer_data_y = str(data.get('accerometer_data_y')),accerometer_data_z = str(data.get('accerometer_data_z')),gyroscope_data_x = str(data.get('gyroscope_data_x')),gyroscope_data_y = str(data.get('gyroscope_data_y')),gyroscope_data_z = str(data.get('gyroscope_data_z')),optical_sensor_data = str(data.get('optical_sensor_data')),device_id = str(data.get('device_id')),geo_location =str(data.get('latitude'))+','+str(data.get('longitude')))

### What I Did
Trying to connect Bleak with GCP IoT
```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

hbldh, 2019-05-25:
I don't understand what you mean is the problem? The code you provided does not contain any bleak-related code and no indication of what has gone wrong. 
If you have a problem sending data to GCP IoT, then you should ask someone else than me. If the problem is communicating with your device then state what you did with your device, not your GCP connection.

amitlalaawaz, 2019-05-28:
Thank you for your response. We are looking to upload data to GCP IoT, but nowhere we are getting clue or hint to upload

hbldh, 2019-06-24:
I will close this issue. You are welcome to open a new issue if you have further problems with Bleak.

closed 2019-06-24 by hbldh