hbldh/bleak #69: Unhandled error in discovery()

corentinbettiol, 2019-06-03:
* bleak version: 0.4.2
* Python version: 3.6.7
* Operating System: Xubuntu 18.04 (4.15.0-50-generic)

### Description

Every time I try to list BLE devices (using `discover()`), bleak lists the devices and prints 20 to 50 times the same beautiful unhandled error.

It seems to appear randomly (1-2 times every 2-5 try).

The function is [here](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/discovery.py#L38).
[image]

### What I Did

Here's the error:

```
Unhandled Error
Traceback (most recent call last):
  File "/home/usr/dir/virtualEnv/lib/python3.6/site-packages/txdbus/protocol.py", line 141, in dataReceived
    self.rawDBusMessageReceived(raw_msg)
  File "/home/usr/dir/virtualEnv/lib/python3.6/site-packages/txdbus/protocol.py", line 262, in rawDBusMessageReceived
    self.signalReceived(m)
  File "/home/usr/dir/virtualEnv/lib/python3.6/site-packages/txdbus/client.py", line 643, in signalReceived
    self.router.routeMessage(msig)
  File "/home/usr/dir/virtualEnv/lib/python3.6/site-packages/txdbus/router.py", line 128, in routeMessage
    r.match(m)
--- <exception caught here> ---
  File "/home/usr/dir/virtualEnv/lib/python3.6/site-packages/txdbus/router.py", line 67, in match
    self.callback(m)
  File "/home/usr/dir/virtualEnv/lib/python3.6/site-packages/bleak/backends/bluezdbus/discovery.py", line 109, in parse_msg
    *_device_info(msg_path, devices.get(msg_path))
  File "/home/usr/dir/virtualEnv/lib/python3.6/site-packages/bleak/backends/bluezdbus/discovery.py", line 39, in _device_info
    name = props.get("Name", props.get("Alias", path.split("/")[-1]))
builtins.AttributeError: 'NoneType' object has no attribute 'get'
```

corentinbettiol, 2019-06-04:
Super-quick workaround is testing `props` and returning something (like `0,0,0,0`) if `props` is `False`.

In `backends/bluezdbus/discovery.py`, [line 38](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/discovery.py#L38):
```python
def _device_info(path, props):
    if props:
        name = props.get("Name", props.get("Alias", path.split("/")[-1]))
        address = props.get("Address", None)
        if address is None:
            try:
                address = path[-17:].replace("_", ":")
                if not validate_mac_address(address):
                    address = None
            except Exception:
                address = None
        rssi = props.get("RSSI", "?")
        return name, address, rssi, path
    return 0,0,0,0```

hbldh, 2019-06-05:
Thank you for this issue. I will evaluate this as soon as possible.

hbldh, 2019-06-24:
This will be addressed in version 0.4.3 which will be released shortly.

closed 2019-06-30 by hbldh

hbldh, 2019-07-01:
Released 0.4.3 yesterday. This is handled better there. Reopen issue if it persists.

jollierfinbean, 2019-08-20:
This still seems to happen
```
 $ pip freeze|grep bleak
bleak==0.5.0
```
```
$ python --version
Python 3.7.3
```
```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.3 LTS"
```
```
# test.py
import logging

import asyncio
from bleak import discover

logging.basicConfig(
    level=logging.INFO,
    format='%(levelname)7s: %(message)s',
)
log = logging.getLogger('asyncio')

async def run():
    devices = await discover()
    for d in devices:
        pass

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```

output:
```
$ python test.py
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-64 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-67 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 63-CA-B7-02-E3-6A, 63:CA:B7:02:E3:6A (-85 dBm), Object Path: /org/bluez/hci0/dev_63_CA_B7_02_E3_6A
   INFO: F2-50-A2-35-BB-5E, F2:50:A2:35:BB:5E (-92 dBm), Object Path: /org/bluez/hci0/dev_F2_50_A2_35_BB_5E
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-69 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-77 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-69 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 63-CA-B7-02-E3-6A, 63:CA:B7:02:E3:6A (-96 dBm), Object Path: /org/bluez/hci0/dev_63_CA_B7_02_E3_6A
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-69 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: D8-8C-2E-82-B3-43, D8:8C:2E:82:B3:43 (-85 dBm), Object Path: /org/bluez/hci0/dev_D8_8C_2E_82_B3_43
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-65 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-70 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-70 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-78 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-71 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-65 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 63-CA-B7-02-E3-6A, 63:CA:B7:02:E3:6A (-86 dBm), Object Path: /org/bluez/hci0/dev_63_CA_B7_02_E3_6A
   INFO: D8-8C-2E-82-B3-43, D8:8C:2E:82:B3:43 (-84 dBm), Object Path: /org/bluez/hci0/dev_D8_8C_2E_82_B3_43
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-72 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-78 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 63-CA-B7-02-E3-6A, 63:CA:B7:02:E3:6A (-86 dBm), Object Path: /org/bluez/hci0/dev_63_CA_B7_02_E3_6A
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-71 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
[... 31 lines cut ...]
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-65 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-71 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: D8-8C-2E-82-B3-43, D8:8C:2E:82:B3:43 (-96 dBm), Object Path: /org/bluez/hci0/dev_D8_8C_2E_82_B3_43
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-65 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC
   INFO: F2-50-A2-35-BB-5E, F2:50:A2:35:BB:5E (-92 dBm), Object Path: /org/bluez/hci0/dev_F2_50_A2_35_BB_5E
   INFO: InterfacesRemoved, org.freedesktop.DBus.ObjectManager (/): ['/org/bluez/hci0/dev_F2_50_A2_35_BB_5E', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
  ERROR: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
  File "/home/samuli/Projektit/be/venv/lib/python3.7/site-packages/bleak/backends/bluezdbus/discovery.py", line 40, in _device_info
    name = props.get("Name", props.get("Alias", path.split("/")[-1]))
AttributeError: 'NoneType' object has no attribute 'get'
   INFO: None, None (None dBm), Object Path: None
   INFO: D8-8C-2E-82-B3-43, D8:8C:2E:82:B3:43 (-96 dBm), Object Path: /org/bluez/hci0/dev_D8_8C_2E_82_B3_43
   INFO: InterfacesRemoved, org.freedesktop.DBus.ObjectManager (/): ['/org/bluez/hci0/dev_D8_8C_2E_82_B3_43', ['org.freedesktop.DBus.Properties', 'org.freedesktop.DBus.Introspectable', 'org.bluez.Device1']]
  ERROR: 'NoneType' object has no attribute 'get'
Traceback (most recent call last):
[... cut ...]
AttributeError: 'NoneType' object has no attribute 'get'
   INFO: None, None (None dBm), Object Path: None
   INFO: 63-CA-B7-02-E3-6A, 63:CA:B7:02:E3:6A (-86 dBm), Object Path: /org/bluez/hci0/dev_63_CA_B7_02_E3_6A
   INFO: 40-F5-10-A8-08-4C, 40:F5:10:A8:08:4C (-59 dBm), Object Path: /org/bluez/hci0/dev_40_F5_10_A8_08_4C
   INFO: 59-54-2A-67-B4-17, 59:54:2A:67:B4:17 (-71 dBm), Object Path: /org/bluez/hci0/dev_59_54_2A_67_B4_17
   INFO: 70-73-CB-D5-70-DC, 70:73:CB:D5:70:DC (-65 dBm), Object Path: /org/bluez/hci0/dev_70_73_CB_D5_70_DC

hbldh, 2019-09-07:
@jollierfinbean The traceback of the error is only printed now, instead of the application crashing is it not? 
The printout will disappear as well in version 0.5.1, btw.

jollierfinbean, 2019-09-08:
> @jollierfinbean The traceback of the error is only printed now, instead of the application crashing is it not?
> The printout will disappear as well in version 0.5.1, btw.

Yes, it does not crash. Also thank you for removing the exception logging since it cluttered my logs pretty badly. :+1:

commit 708f03d3 (2019-06-24): Fixes #69
commit 9aebbb28 (2019-06-30): Version 0.4.3 - Multiple bugfixes

Merge branch 'release/v0.4.3'
Fixes #76 - Using other adapter than hci0
Fixes #75 - Timeout for discover in connect, .NET
Fixes #74 - Connection unsuccessful reported
Fixes #70 - Problem connection in Windows
Fixes #69 - BlueZ discovery bug
Fixes #68 - Dependabot
Fixes #78 - Bleak Logo
Added RSSI property to BLEDevices returned by discovery methods
Modified `details` field to BlueZ BLEDEvices
Documentation updates