hbldh/bleak #70: Problem connecting to BLE devices with Windows

corentinbettiol, 2019-06-05:
* bleak version: 0.4.2
* Python version: 3.7.3
* Operating System: Windows 10

### Description

On linux (`backends/bluezdbus/client.py`), `connect()` function uses `discover()` with a timeout : `await discover(timeout=0.1, loop=self.loop)` ([here](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/client.py#L64)).

On windows (`backends/dotnet/client.py`), `connect()` function uses `discover()` with a defined value: `devices = await discover(2.0, loop=self.loop)` ([here](https://github.com/hbldh/bleak/blob/master/bleak/backends/dotnet/client.py#L89)), which leads to random errors due to the fact that Windows is unable to discover the device to connect within 2 seconds.

Since `discover()` [has a timeout var](https://github.com/hbldh/bleak/blob/master/bleak/backends/dotnet/discovery.py#L28), it can be good to use `timeout=2.0` instead of `2.0` in `client.py`, so developers can adapt the value without having to manually change Bleak.

hbldh, 2019-06-05:
Thank you for your ideas. I agree, I have thought about this as well. My initial idea had been to not have any keyword argument at all for the `connect` method, but I think I might have to reconsider this. 
I will probably make a temporary fix in version 0.4.3 to give users the configurability you ask for, but I have an idea for a more permanent solution which requires a lot more time to develop; time I do not have right now I am afraid. I will get back to you here when I have made some modifications and/or plans.

corentinbettiol, 2019-06-06:
Thank you for your quick answer, I will look at v0.4.3 when it will be available:)

closed 2019-06-30 by hbldh

hbldh, 2019-07-01:
Released 0.4.3 yesterday. Timeout keyword is now available in both BlueZ and .NET backend.

sonus89, 2019-09-07:
I have a similar problem. 
Windows 10 + Python 3.7 + bleak
I can not even import BleakClient from bleak
I also tried bleson with no result...
I can only discover / scan / list nearby devices but unable to connect.

Any idea?

hbldh, 2019-09-07:
@sonus89 please create a new issue and add code and traceback of errors there.

commit 9aebbb28 (2019-06-30): Version 0.4.3 - Multiple bugfixes

Merge branch 'release/v0.4.3'
Fixes #76 - Using other adapter than hci0
Fixes #75 - Timeout for discover in connect, .NET
Fixes #74 - Connection unsuccessful reported
Fixes #70 - Problem connection in Windows
Fixes #69 - BlueZ discovery bug
Fixes #68 - Dependabot
Fixes #78 - Bleak Logo
Added RSSI property to BLEDevices returned by discovery methods
Modified `details` field to BlueZ BLEDEvices
Documentation updates