hbldh/bleak #71: Problem with function client.write_gatt_char()

Szymon-Gesicki, 2019-06-06:
* bleak version: 0.4.2
* Python version: 3.7
* Operating System: win 10

I am trying to communicate to device by bluetooth. I have the UUID, also I'm doing everything that you showed in issue #59 but no matter what I'm typing to 'data' in function client.write_gatt_char  it shows me this type of error:
"Could not write value 'data' to characteristic"
I've tried bytes or bytearray, but everything ends with the same error.
I decided to comment this function. Unfortunately, the function callback has never been called.
My code:
[example.txt](https://github.com/hbldh/bleak/files/3260721/example.txt)

closed 2019-06-12 by Szymon-Gesicki

Surgavery, 2019-06-25:
Hi Szymon-Gesicki,

I have similar problem when using version 0.4.2. I suggest you try to downgrade the bleak version to 0.4.0.

Huygon, 2023-10-23:
How do you solve your problem @Szymon-Gesicki ? thanks in advance