hbldh/bleak #72: Wake up via BLE

Jeins98, 2019-06-11:
* bleak version: 0.4.0
* Python version: 3.6
* Operating System: Windows/Ubuntu

I want to try to wake up my devices via ble. On windows it works perfectly, when my Advertising Intervall is very long. But on Ubuntu it doesn't work. It works when I reduce my Intervall.
You have a fix for me?

Regards

hbldh, 2019-06-24:
No, I do not have a fix for you. 

What do you mean? Does it work on Ubuntu when you reduce your interval? 
What Advertising Interval are you refering to? Is it a peripheral that you have written yourself and control the interval of?

As long as the device is not in deep sleep but is still advertising, then it should be possible to wake up/connect and talk to. Try running a long discover before desicovery on Ubuntu to get the BlueZ DBUS API to store the device in its registers.

Jeins98, 2019-07-03:
On Windows I can do a read gatt or notify and after a period of time my sensor wake up (out of deep sleep) and I get the data. My Advertise Intervall now is between 3000-5000 ms. But I want to extend the intervall to save battery. With Windows Intervalls between 5000-7000 ms work's fine. Here the problem is linux. There I can't connect to the sensor.

hbldh, 2019-07-04:
Is the problem that you do not even find the device when running the `discover` method? 

Or do you create a client, make the connection and then reuse the same client after a long wait? BlueZ might terminate the connection for you if it is unused for a long time. Such effects are something I know very little about right now...

I need a clearer description of the workflow of your application to be able to pinpoint anything else.

Otherwise, in version 0.4.3 there is a new keyword `timeout` which you can send to the `connect` method, making bleak running an internal `discover` for a longer time. It might be that the short `discover` which is run by default (0.1 s) is to small to pick up your advertisement...

Jeins98, 2019-07-04:
First thanks for your awnser. I have a sensor, from which I know the MAC-Adress. And I have a server who try to connect every 5 min to the sensor to get the data. I only create a client on the known MAC and try to connect. On Windows it takes some time to connect but it worked. On Linux it only told me "can't connect".

hbldh, 2019-10-09:
If you are still experience problems with this, install the new version 0.5.1. and try doing like this instead:

```python
import asyncio
import platform

from bleak import BleakClient

async def connect_and_do_your_thing(mac_addr: str, loop: asyncio.AbstractEventLoop):
    async with BleakClient(mac_addr, timeout=10.0, loop=loop) as client:
        # Do your thing here.
        svcs = await client.get_services()
        print("Services:", svcs)

mac_addr = (
    "24:71:89:cc:09:05"
    if platform.system() != "Darwin"
    else "243E23AE-4A99-406C-B317-18F1BD7B4CBE"
)
loop = asyncio.get_event_loop()
loop.run_until_complete(connect_and_do_your_thing(mac_addr, loop))
```
The timeout added to the client constructor might remove your issues.

hbldh, 2020-09-25:
This might be solved by version 0.8.0. Will close for now.

closed 2020-09-25 by hbldh