hbldh/bleak #73: BLE devices names inconsistent between Linux & Windows

corentinbettiol, 2019-06-12:
* bleak version: 0.4.2
* Python version: 3.7.3
* Operating System: Windows 10 & Xubuntu 19.04

### Description

When launching discover function on Linux, we get a list of devices. And if the device name is unknown, it is replaced by its address.
On Windows however, the names are just "Unknown".
Will the output be the same in the next version ? Or is it a normal behavior ?

[image]

The Windows "Unknown" problem seems to be located here:

https://github.com/hbldh/bleak/blob/4f3b890e8f132fc2cd48d1588a372b152049a5fc/bleak/backends/dotnet/discovery.py#L57

hbldh, 2019-06-24:
It is not a problem per se, but an effect of how the OS specific systems work. 

If there is no `Name` property in Ubuntu/BlueZ I use the `Alias` property:
https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/discovery.py#L39
This is as good as it gets there; if no `Alias` exists wither, than `Unknown` is set.

On Windows on the other hand, there is nothing else than `LocalName` and if it does not exist there is nothing else I can do.

For the `discovery` method I aim to be consistent on the platform, so that discovery via other OS native methods give similar results to Bleak gives.

dlech, 2019-06-24:
I observed the same thing on Windows. Even though the device has the proper advertising data (confirmed by using nRF Connect on Android), the Windows Bluetooth stack does not seem to get the name from this for some reason.

hbldh, 2019-10-09:
I do not think I can do anything more in regards to this issue. Closing it.

closed 2019-10-09 by hbldh

dlech, 2019-10-09:
I was wondering if #99 might fix this, but I haven't had a chance to try yet.

hbldh, 2019-10-09:
I have little hope of fixing the difference problem, but I think I see an improvement in the Windows discovery, albeit not enough that I feel comfortable saying that this is fixed.