hbldh/bleak #74: Connection was not successful, however it was!

skjerns, 2019-06-12:
* bleak version: 0.4.2
* Python version: 3.7
* Operating System: Win10

### Description

I'm wrestling with the problem that I cannot connect to a BLE device on Windows. Sometimes it works, and sometimes it doesn't, and I have not found any clue to what influences it.

However, I have seen that the Windows-Bluetooth-Indicator (Tray-Symbol) changes to 'Connected (1)' for a second, after which `bleak` tells me 'BleakError("Connection to {0} was not successful!"'.

Now I have tried bridging the line 118 in `client.py`:
```Python
#connected = await client.is_connected()
connected = True
```

So I think that we might need to increase the `await asyncio.sleep(0.2, loop=self.loop)` to a higher value (500ms?). This also fixed my issue.

Do you have any idea what is going wrong with the `client.is_connected`? Does it just need more time? Or is there a better way to check if we are connected?

hbldh, 2019-06-24:
This is strange. If the `get_services` call on the line before the 0.2 s sleep is successful, then we are connected. What kind of device are you connecting to?

If you change to `connected=True` and call `await client.is_connected()` after connection, does it still return `False`?

I will change this in version 0.4.3 and wil notify you here when I upload that version.

skjerns, 2019-06-26:
It's a custom device / pcb we are creating ourselves, using an off-the-shelf BLE module.

I injected the `connected=True` in the `dotnet/client.py/connect()` and a subsequent call to `client.is_connected()` yields True as well.
 
```Python
import bleak
import asyncio

async def test():
    address = 'd8:80:39:fa:7a:b2' # P5
    client = bleak.BleakClient(address.upper())
     
    await client.connect() # with connected=True injected
    print(await client.is_connected()) # -> prints True
    return
    
loop = asyncio.get_event_loop()
loop.run_until_complete(test())
```

This even returns True when I remove the `await asyncio.sleep(0.2, loop=self.loop)`, however then it sometimes gives me a `Could not get GATT services.`

Very strange behaviour indeed.

closed 2019-06-30 by hbldh

hbldh, 2019-07-01:
Released 0.4.3 yesterday, including a change for handling [this problem](https://github.com/hbldh/bleak/blob/master/bleak/backends/dotnet/client.py#L137-L146).

Reopen the issue if the problem comes back.

commit 9aebbb28 (2019-06-30): Version 0.4.3 - Multiple bugfixes

Merge branch 'release/v0.4.3'
Fixes #76 - Using other adapter than hci0
Fixes #75 - Timeout for discover in connect, .NET
Fixes #74 - Connection unsuccessful reported
Fixes #70 - Problem connection in Windows
Fixes #69 - BlueZ discovery bug
Fixes #68 - Dependabot
Fixes #78 - Bleak Logo
Added RSSI property to BLEDevices returned by discovery methods
Modified `details` field to BlueZ BLEDEvices
Documentation updates