hbldh/bleak #75: Timeout for discover in connect()

skjerns, 2019-06-13:
* bleak version: 0.4.2
* Python version: 3.7
* Operating System: Win10

### Description

When connecting via DotNet backend, `discover` is called with a timeout of `2.0`. However, this timeout is not always sufficient to get responses from the device necessary. In around 1/3rd of the cases it will not be able to connect to my device due to not finding it within 2 seconds.

Do you think that here a 'continue looking for device until found or timeout' loop would be useful? So the maximal timeout would be the default `5.0`, but it can be shorter if the device is found before. Else, the connect method might have a parameter `force` that also tries to connect even if the device was not found during radio?

hbldh, 2019-06-24:
Duplicate of #70.
This is fixed in  6daae16 and will be released in version 0.4.3 shortly.

closed 2019-06-30 by hbldh

hbldh, 2019-07-01:
Released 0.4.3 yesterday. Timeout keyword is now available in both BlueZ and .NET backend.

commit 9aebbb28 (2019-06-30): Version 0.4.3 - Multiple bugfixes

Merge branch 'release/v0.4.3'
Fixes #76 - Using other adapter than hci0
Fixes #75 - Timeout for discover in connect, .NET
Fixes #74 - Connection unsuccessful reported
Fixes #70 - Problem connection in Windows
Fixes #69 - BlueZ discovery bug
Fixes #68 - Dependabot
Fixes #78 - Bleak Logo
Added RSSI property to BLEDevices returned by discovery methods
Modified `details` field to BlueZ BLEDEvices
Documentation updates