hbldh/bleak #76: Using adapter other than 'hci0' gives error during discovery

yggdr, 2019-06-19:
* bleak version:
0.4.2 + workaround patch from #69 (+ a few `print`s for debugging, so line numbers might be off by 1 or 2)
* Python version:
3.7.3
* Operating System:
Ubuntu 19.04 / `Linux 5.0.0-16-generic #17`

### Description

Using an adapter other than 'hci0' does get set in the `BleakClientBlueZDBus` instance, but not propagated to all called functions, like e.g. `backends.bluezdbus.discovery.discover`

### What I Did

Create an instance with a non-default device and have it try to connect. Very stripped down version of our code:

```
from bleak import BleakClient
import sys
import asyncio

class MyInternalStuff:
    def __init__(self, ble_addr, *, adapter="hci0", loop):
        self._adapter = adapter
        self._dev = BleakClient(ble_addr, device=self._adapter, timeout=35)
    async def __aenter__(self):
        await self._dev.__aenter__()
    async def __aexit__(self, exc_type, exc_val, exc_tb):
        await self._dev.__aexit__(exc_type, exc_val, exc_tb)

async def main(loop):
    async with MyInternalStuff(ble_addr="12:34:56:78:90:ab", adapter='hci1', loop=loop):
        pass

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    sys.exit(loop.run_until_complete(main(loop)))
```

gives 

```
Traceback (most recent call last):
  File "bug.py", line 25, in <module>
    sys.exit(loop.run_until_complete(main(loop)))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "bug.py", line 19, in main
    async with MyInternalStuff(ble_addr="12:34:56:78:90:ab", adapter='hci1', loop=loop):
  File "bug.py", line 12, in __aenter__
    await self._dev.__aenter__()
  File "/home/yggdrasil/.virtualenvs/svhsystemtests/lib/python3.7/site-packages/bleak/backends/client.py", line 41, in __aenter__
    await self.connect()
  File "/home/yggdrasil/.virtualenvs/svhsystemtests/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 65, in connect
    await discover(timeout=0.1, loop=self.loop)
  File "/home/yggdrasil/.virtualenvs/svhsystemtests/lib/python3.7/site-packages/bleak/backends/bluezdbus/discovery.py", line 143, in discover
    adapter_path, interface = _filter_on_adapter(objects, device)
  File "/home/yggdrasil/.virtualenvs/svhsystemtests/lib/python3.7/site-packages/bleak/backends/bluezdbus/discovery.py", line 26, in _filter_on_adapter
    raise Exception("Bluetooth adapter not found")
Exception: Bluetooth adapter not found
```

Patching  `backends/bluezdbus/client.py:65` like so:
`await discover(timeout=0.1, device=self.device, loop=self.loop)`
fixes this particular instance, but there might be more places this is missing.

hbldh, 2019-06-24:
This is definitely a an oversight, and I will remedy this bug in version 0.4.3, which will be released shortly.

closed 2019-06-30 by hbldh

hbldh, 2019-07-01:
Made necessary modifications, tested it with a second adapter in Ubuntu 18.04 and released 0.4.3. 
Thank you for reporting this and reopen if you find further problems with this.

commit 9aebbb28 (2019-06-30): Version 0.4.3 - Multiple bugfixes

Merge branch 'release/v0.4.3'
Fixes #76 - Using other adapter than hci0
Fixes #75 - Timeout for discover in connect, .NET
Fixes #74 - Connection unsuccessful reported
Fixes #70 - Problem connection in Windows
Fixes #69 - BlueZ discovery bug
Fixes #68 - Dependabot
Fixes #78 - Bleak Logo
Added RSSI property to BLEDevices returned by discovery methods
Modified `details` field to BlueZ BLEDEvices
Documentation updates