hbldh/bleak #77: Cannot set LE security level

yggdr, 2019-06-19:
* bleak version: 0.4.2
* Python version: 3.7.3
* Operating System: Ubuntu 19.04

### Description

Cannot set BTLE security level before connecting.

I'm currently rewriting an internal testing tool from bluepy to an async library. I decided to test out bleak, but one of the problems I ran into was our device giving an "LE security request" when connecting to it. This didn't happen with the old library.
After some talking with our Bluetooth guys and testing with `gatttool` it seems that the old library defaults to what gatttool would call `sec-level low`, while bleak seems to default to at least medium. `bluepy` allows setting this, but I cannot find anything in the `bleak` API description on how to set the security level, so I looked at the bluez dbus "documentation", hoping I could hack in a workaround, but couldn't find anything either.

Any help is appreciated.

hbldh, 2019-06-24:
This is something I definitely want to support, but I actually do not know how. I ahve received multiple questions about security related reading and writing issues with Bleak, but I have failed to find much documentation about it.

There is [https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt#n180](a read and write property) on the `org.bluez.Device1` interface in the BlueZ D-Bus API. Try to set this to `True` and see if that changes anything. Setting might be done with something similar as to this:
```python
from bleak.backends.bluezdbus import defs
await client._bus.callRemote(
    client._device_path,
    "Set",
    interface=defs.PROPERTIES_INTERFACE,
    destination=defs.BLUEZ_SERVICE,
    signature="ssv",
    body=[defs.DEVICE_INTERFACE, "Trusted", True],
    returnSignature="a{sv}",
).asFuture(self.loop)
```
Try setting both before and after `connect` (in the before case, use the `get_device_object_path("hci0", "XX:XX...")` method to get the device path)

See documentation about the `org.freedesktop.DBus.Properties` interface [here](https://dbus.freedesktop.org/doc/dbus-specification.html)

I have no devices to test this on myself, so you will have to help me with this.

yggdr, 2019-06-24:
Thanks for the response. Putting this before the `connect` yields an `AttributeError: 'NoneType' object has no attribute 'callRemote'`, putting it after doesn't change the original outcome :(

hbldh, 2019-06-24:
Ok. Thank you very much for testing the code; it was a bit of a leap of faith...

There is something I need to focus on in the handling of secure BLE connections apparently. Would you mind telling me what kind of device you are trying to connect to so I can buy one and try to get Bleak to work against it and by doing so hopefully all other BLE peripherals with additional security requirements? Or another BLE device, which costs as little as possible but still exibits this behaviour if you know of that?

hbldh, 2019-06-24:
Possibly related to #64.

yggdr, 2019-06-24:
The device is not out yet, so unfortunately I can't help you there :( I'm also not really a user of Bluetooth (ironic, I know), so I don't know any specific device that I can recommend here. I think it should be enough to take any BT device that can be set to not require a PIN code to pair to have the same effect.

hbldh, 2019-06-25:
Ok, I will continue my search for devices that I can use for testing. Until then, I cannot do much more than that.

stevennyman, 2019-06-27:
This Stack Overflow answer explains that elevated security levels are required for pairing iPhones, so those might be testing devices. https://stackoverflow.com/a/24725691

vilvo, 2019-10-04:
I'm interested in the same but for Windows. 
In the first phase we'd like to automate pairing with our BLE device for an internal testing tool implemented with Bleak.
I found there's this from Microsoft:
https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/DeviceEnumerationAndPairing
So I read it should be possible.
Would you like another issue to track this request?

hbldh, 2019-10-08:
If it is possible then you are more than welcome to do the work and send me a Pull Request. 
It is not a problem as such to me since I have no interest in connecting to protected devices right now, so you commenting on this issue are more suited to perform the implementation and testing of this issue.

andrewleech, 2019-10-23:
I've been trying to get pairing / bonding to work on windows without much luck - running into issues/limitation with pythonnet (and/or my understanding of how to use it).

I've tried discussing it with them without much resolution: https://github.com/pythonnet/pythonnet/issues/940

hbldh, 2020-09-25:
Pairing is implemented for .NET and BlueZ backends in version 0.8.0. Will close this for now.

PR #269 (2020-09-02): First drafts of "Just Works" pairing for .NET and BlueZ backends
Might fix #64
Might fix #77
Might fix #203 
Fixes #218
closed 2020-09-25 by hbldh