hbldh/bleak #81: BleakServer

kevincar, 2019-07-05:
Currently, bleak establishes a BleakClient that can scan for BLE peripheral devices. 

Are there any future plans to implement a BleakServer to establish a central device that can advertise services and characteristics that other devices can connect to?

I imagine a BleakServer could utilize `BleakGATTService` and `BleakGATTCharacteristic` objects to set up and manage user-defined services that are then advertised and could be read from, written and/or subscribed to.

hbldh, 2019-07-08:
It would be nice but there are no immediate plans; I have had ideas about a Flask-like decorator approach to adding services and characeristics. It does however require a renaming of this package since it then no loger is a Platform agnostic client but also a server...
My focus have been set on getting the macOS backend in place first; but now that it has been almost released, I have to find a new milestone for bleak...

filips123, 2019-10-03:
@hbldh Any update? Bleak is an amazing cross-platform BLE client. However, BLE server (possibly also cross-platform) would be another amazing and needed addition.

kevincar, 2019-10-05:
@filips123 Until there are future plans to implement a BLE server into bleak, some colleagues and I implement a [rough draft of a server](https://github.com/kevincar/bleak/tree/server) for bleak. We'v tested it on both Windows and OS X platforms and it works for our purposes (no Linux yet). Hopefully something similar can be officially added to bleak in the future; although as it has been noted, in this case, bleak is no longer a client only package.

hbldh, 2019-10-07:
The main reason for not implementing a server part in Bleak is that I as a developer has no personal usecase for it. I am reluctant to add that functionality to Bleak since I am struggling to maintain this project as it is, and it would be detrimental to its health to add another part for which I have no personal interest. 
It would be much better if someone else would take the lessons learned in this project and make a standalone server-package, much like @kevincar has done. Everyone would benefit from that instead of adding it here, I think.

hbldh, 2020-09-25:
Closing this since it is outside of the Bleak project scope.

closed 2020-09-25 by hbldh

kevincar, 2020-12-15:
As a reference, and for those looking for a potential cross-platform python BLE peripheral solution, I decided to create a [companion project](https://github.com/kevincar/bless) from that rough draft previously mentioned as a companion to this project.

Thanks for the motivation @hbldh!

cspensky, 2021-01-22:
@kevincar I'd love to help out with bless and get this functionality stable.

kevincar, 2021-01-23:
@cspensky Certainly! Feel free to fork it, submit pull requests, open issues, or begin [discussions](https://github.com/kevincar/bless/discussions) there.