hbldh/bleak #82: Connection handling enhancements

JoeHut, 2019-07-07:
In Bluez it should be possible to listen for an event if a device has disconnected (https://stackoverflow.com/questions/53664144/device-connect-disconnect-notification-via-bluez-dbus-api). Are there plans to add an disconnection callback to the client API (I have no idea whether this is possible on the other platforms) or even try to implement something like auto-reconnection in the library itself?

hbldh, 2019-07-08:
There are no plans, no. It is a good idea though and I do not think it is very hard to do, at least not in the BlueZ backend. 

The auto-reconnection is somewhat breaking the idea of BLE, is it not? If the connection is not used, better to let it die and let the peripheral save energy until it is actually needed and let the client reconnect then?

JoeHut, 2019-07-09:
I could provide a PR for the API and a BlueZ implementation, but I have no idea about .NET or the Darwin backend...

Out of experience there are multiple reasons why a BLE connection might drop, especially if the connection intervals are not properly set up. Typical cases are interferences and reflections in crowded environments and getting out of range. We always try to have a very resilient auto-reconnection on the peripheral side since you typically want to have long-lived connections where you can react to events from the peripheral in real time. But this is of course very use-case specific.

Anyways, this logic can be definitely pushed to the application if we have a proper disconnect callback.

hbldh, 2019-08-02:
This is implemented and merged into version 0.5.0 which has now been deployed to PyPI.

PR #86 (2019-07-21): Register a disconnected callback and clean up subscriptions on disconnect
This patch contains two changes:
* Add API and Bluez implementation to set a callback fired when a device disconnects without active intervention from the client
* Remove all existing subscriptions to characteristic notifications when a device disconnects

Replaces #85 
Fixes #83 
Closes #82
closed 2019-08-02 by hbldh