hbldh/bleak #87: "dictionary changed size during iteration" with .NET discover()

hexaguin, 2019-07-15:
* bleak version: 0.4.3
* Python version: 3.7.3 (Anaconda)
* Operating System: Windows 10 (build 1903)

### Description

Very infrequently (i.e. once every few hours of continuous scanning), running the `discover()` function will cause a runtime error. It would appear that the internal dictionary `devices` is being modified while Bleak iterates over it.

It's easy enough to work around (for now I'm catching this particular exception and re-running `discover()`), but this bug is irritating, to say the least. Sometimes it happens within a few cycles of calling `discover()`, sometimes it won't show up for a few hours at a time.

### What I Did

1. Make a program that repeatedly calls `discover()`
2. Wait for the following traceback:

```
Traceback (most recent call last):
  File "[my script].py", line 140, in <module>
    loop.run_until_complete(scan_loop())
  File "[My user folder]\Anaconda3\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "[my script].py", line 102, in scan_loop
    devices = await discover(device="hci0", timeout=3)
  File "[My user folder]\lib\site-packages\bleak\backends\dotnet\discovery.py", line 91, in discover
    for d in devices.values():
RuntimeError: dictionary changed size during iteration
```

hbldh, 2019-07-16:
I have never tested a `few hours of continuous scanning`, so this was a surprise. I guess there is some potential for writing during iteration. I will check this out eventually.

jckruege, 2020-02-17:
I'm experiencing this issue with v0.5.1,  using first example on https://pypi.org/project/bleak/

jckruege, 2020-02-17:
I manually added JoeHut change to bleak/backends/dotnet/discovery.py (I have no clue what I'm doing), and now no crash. Thank you.

nicosoto0, 2020-03-02:
I also have still the same problem.  I have a application that haves to do continuous scanning, and after 9 hours  (aprox.) I fall in the following exception:
`dictionary changed size during iteration`
Even if I capture the exception, after it happens ones it start happening more often (once every 30 minutes aprox.) and after a couple of catches the program stops because of:
`Process is terminated due to StackOverflowException.`

commit f5d86a04 (2020-01-05): Fixes #87.
commit 32aa57b9 (2020-03-09): Fixes #87.
PR #164 (2020-03-09): Release/v0.6.0
* New Scanner object to allow for async device scanning.
* Updated ``txdbus`` requirement to version 1.1.1 (Merged #122)
* Implemented ``write_gatt_descriptor`` for Bluez backend.
* Large change in Bluez backend handling of Twisted reactors. Fixes #143
* Modified ``set_disconnect_callback`` to actually call the callback as a callback. Fixes #108.
* Added another required parameter to disconnect callbacks.
* Added Discovery filter option in BlueZ backend (Merged #124)
* Merge #138: comments about Bluez version check.
* Improved scanning data for macOS backend. Merge #126.
* Merge [... cut]
closed 2020-03-09 by hbldh