hbldh/bleak #88: Write size limited to 127 bytes?

HenryHu, 2019-07-16:
* bleak version: 0.4.3
* Python version: 3.7.3
* Operating System: Windows 10

### Description

I'm trying to use bleak to write to a BLE characteristic.
If I write with a buffer shorter or equal to 127 bytes, the write succeeds, and the device behaves as expected.
However, if I write a buffer longer than 127 bytes, I got an exception.
Is there a length limit on the write_gatt_char() method?

### What I Did

  File "C:\Tools\Python\lib\site-packages\bleak\backends\dotnet\client.py", line 372, in write_gatt_char
    loop=self.loop,
  File "C:\Tools\Python\lib\site-packages\bleak\backends\dotnet\utils.py", line 73, in wrap_IAsyncOperation
    raise BleakDotNetTaskError(op.ErrorCode.ToString())
bleak.exc.BleakDotNetTaskError: System.ArgumentException: Value does not fall within the expected range.

hbldh, 2019-07-16:
I had not encountered this myself. Can you test if it is the same using bleak with the BlueZ backend?

Generally, A GATT characteristic should be able to handle up to 512 bytes of data in one write, I think. It might be an issue due to automatic conversions in `pythonnet`. I will try this out in a while.

hbldh, 2019-11-02:
https://stackoverflow.com/questions/53313117/cannot-write-large-byte-array-to-a-ble-device-using-uwp-apis-e-g-write-value

I have evaluated this and I have not reached any other hypothesis than that the peripherial itself limits the size of what can be written. This does however not seem to be available in the UWP Bluetooth API.

closed 2019-11-02 by hbldh