hbldh/bleak #89: Memory leaks through missing callback deregistration in Bluez

JoeHut, 2019-07-16:
* bleak version: master
* Python version: 3.7.3
* Operating System: Linux
* Bluez version: 5.50

### Description

In my use case I have a long running application that is looking for devices with certain properties (like a name prefix, RSSI etc.) and connects to them. To accomplish this I have an infinite loop that calls discover and checks the list of devices until the proper one is available. Something like:

```python
while True:
    devices = await discover(timeout=1)
    for d in devices:
        if d.name.startswith("PREFIX"):
            return d
```

I observed that if it is running for a while, the discover takes not only around a second but up to multiple minutes.

### Explanation

The issue is that the DBus callbacks in discovery() for InterfacesAdded, InterfacesRemoved and PropertiesChanged are never deleted again. Therefore they pile up and the parse_msg callback is called thousands of times after a couple of minutes.

The issue is therefore similar to #83 where the notifications are sent multiple times after a reconnection. I think it makes sense to go through all Bluez callback registrations and make sure they don't leak.

JoeHut, 2019-07-16:
@hbldh I will add a PR for this. Please let me know if there are other ways to handle those callback de-registrations. I am not an expert for DBus/Bluez

hbldh, 2019-07-16:
This is a very good catch. Very bad thing to miss this. Will integrate it asap.

PR #90 (2019-08-02): Fix leaking of callbacks from DBus on multiple calls to discover()
If the rules are only added to Bluez via DBus but never removed again,
the calls pile up if discover() is called multiple times in one session.
This is especially harmful for long-running applications that connect
and disconnect to the same or different servers over and over. The time
to await discovery() becomes very long (up to minutes) even with smaller
timeouts.

Make sure the callbacks registered are cleaned up before discover() is
left.

Fixes #89
commit e4185095 (2019-09-07): Version 0.5.0

macOS support added (thanks to @kevincar)
Merged #80: macOS development
Merged #90 which fixed #89: Leaking callbacks in BlueZ
Merged #92 which fixed #91, Prevent leaking of DBus connections on discovery
Merged #96: Regex patterns
Merged #86 which fixed #83 and #82
Recovered old .NET discovery method to try for #95
closed 2019-08-02 by hbldh