hbldh/bleak #91: Leak of active DBus connections

JoeHut, 2019-07-16:
* bleak version: master
* Python version: 3.7.3
* Operating System: Linux
* Bluez version: 5.50

### Description

DBus connections are not cleaned up after usage. Multiple calls to discover() result eventually in following error:
```
Traceback (most recent call last):
  File "discover.py", line 29, in <module>
    loop.run_until_complete(run())
  File "/usr/lib64/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "discover.py", line 22, in run
    devices = await discover(timeout=1)
  File "/home/joe/code/proglove/python/bleak/bleak/backends/bluezdbus/discovery.py", line 125, in discover
    bus = await client.connect(reactor, "system").asFuture(loop)
txdbus.error.RemoteError: org.freedesktop.DBus.Error.LimitsExceeded: The maximum number of active connections for UID 1000 has been reached
```

For me this happens at the 246th call to discover()

JoeHut, 2019-07-16:
I would assume this also happens if you just connect to the same device multiple times since the client's ``_bus`` is never disconnected...

hbldh, 2019-07-16:
Same this here as in #92. This is an important find that needs addressing as soon as possible.

PR #92 (2019-08-02): Prevent leaking of DBus connections on discovery
During a call to discover() on the Bluez backend, a DBus connection is
opened, but never closed. If discover() is called multiple times while
the application is running, DBus at some point prevents opening
additional connections.

Make sure the bus connection is closed after discovery.

Fixes #91
commit e4185095 (2019-09-07): Version 0.5.0

macOS support added (thanks to @kevincar)
Merged #80: macOS development
Merged #90 which fixed #89: Leaking callbacks in BlueZ
Merged #92 which fixed #91, Prevent leaking of DBus connections on discovery
Merged #96: Regex patterns
Merged #86 which fixed #83 and #82
Recovered old .NET discovery method to try for #95
closed 2019-08-02 by hbldh