hbldh/bleak #93: Discover not working with new event loops.

Mithrandir2k18, 2019-07-17:
* bleak version: 0.4.3
* Python version: 3.5.3
* Operating System: Debian/Raspian-Stretch

### Description
I want to write a small module that deals with a bluetoothle connection.
Everything works fine until I try to use a new event loop together with bleak.discover().
Please see MWE below for details.

### What I Did

```
import asyncio
from bleak import discover
import logging

async def printer2(loop: asyncio.AbstractEventLoop):
    print("Start Printer2")
    devices = await discover(loop=loop)  # This function gets stuck here.
    for d in devices:
        print(d.name)
    print("Hey")
    return 42

async def printer(loop: asyncio.AbstractEventLoop):  # This function works as expected
    print("Start Printer")
    await asyncio.sleep(5, loop=loop)
    print("Hey")
    return 42

def create():
    loop = asyncio.new_event_loop() #  replacing this with asyncio.get_event_loop() works..
    asyncio.set_event_loop(loop)
    return loop

logging.basicConfig(level=logging.DEBUG)
loop = create()
retval = loop.run_until_complete(printer(loop))
print(retval)

retval = loop.run_until_complete(printer2(loop))
print(retval)

```

hbldh, 2019-07-17:
I will have to look into this, but I think it is due to the `twisted` implementation used in `txdbus` package that bleak depends on. Check the [`__init__.py`](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/__init__.py) to see how I create a loop there.
Is it the same you use a new loop for the client class?

djwmarcx, 2019-07-18:
I can confirm this situation in my side. No additional info yet.

Mithrandir2k18, 2019-07-30:
Sorry, didn't have the time to test it right away. Here's a test with my polar h10, but I don't see how a different BLE server would yield different results. Anyway, if the line where create is called is comented out it works as expected, if we create and use a new loop, it won't work. It just waits forever.

```
import asyncio
import bleak
import logging

async def printer2(loop: asyncio.AbstractEventLoop):
    print("Start Printer2")
    devices = await bleak.discover(loop=loop)  # This function gets stuck here.
    for d in devices:
        if "Polar" in d.name:
            return d
    print("Hey")
    return 42

def create():
    loop = asyncio.new_event_loop() #  replacing this with asyncio.get_event_loop() works..
    asyncio.set_event_loop(loop)
    return loop

logging.basicConfig(level=logging.DEBUG)

loop = asyncio.get_event_loop()
retval = loop.run_until_complete(printer2(loop))

loop = create()  # If this line is commented out it workes fine!

client = bleak.BleakClient(address=retval.address, loop=loop)

loop.run_until_complete(client.connect(loop=loop))

print("Connected!")
```

hbldh, 2019-09-07:
The core of the problem here is that bleak uses twisted below asyncio, though the [`asyncioreactor`](https://twistedmatrix.com/documents/current/api/twisted.internet.asyncioreactor.html). Creating a new loop only via asyncio will therefore not work, since it is not the loop installed in the twisted reactor. There might be a way around this, but it involves changing a lot of the code in the BlueZ backend...

The first example you provided works because there is no bleak interaction in the `printer` method. It does not matter whether or not you create a new event loop. The `printer2` however calls `discover` with the new loop, but since it is not the one installed in the twisted reactor, you are waiting on the wrong loop for the results.

The second one is similar. Once you create the new loop and set it as default, all awaiting on the Service bus in bleak will fail.

The simple solution: stop using multiple event loops with bleaks Linux backend.
The hard solution: find a nice way to have a separate reactor for each `BleakClient` and `discover` call. However, this is not a priority for bleak right now...

I will close this in the meantime. If I find the time and a solution I will reopen it.

closed 2019-09-07 by hbldh

Mithrandir2k18, 2019-09-09:
Yeah sounds good. Maybe add this piece of information to the documentation while it's relevant.