hbldh/bleak #95: Device Name Inaccurate

DavidYauUGA, 2019-07-26:
* bleak version: 0.4.3
* Python version: 3.7.4
* Operating System: Win 10

### Description

I was trying to use to discover.py, but they all print out company's name of the bluetooth not the actual setup name for the device. For example, instead of showing "Micheal's Beats3", it will show "Microsoft". 

### What I Did

```
33:FB:D9:92:DE:FA: Microsoft (b'\x01\t \x029\xc5\xc7\xbf\xdb\x88\xe6\xd1\x1e\\\xean\x82U\xb1\xa9Eo\xdc\xf39e9')
```

arasan90, 2019-07-26:
- bleak version: 0.4.3
- Python version: 3.7.4
- Operating System: Win 10

I was about to open a similar issue:
Even if a set a name on the BLE server (on a ESP32 module), I can't see the right name in the "name" field of the discovered device. I tried using an iPhone app and it finds the correct name.

DavidYauUGA, 2019-07-29:
> * bleak version: 0.4.3
> * Python version: 3.7.4
> * Operating System: Win 10
>
> I was about to open a similar issue:
> Even if a set a name on the BLE server (on a ESP32 module), I can't see the right name [...]

I got the same thing except when I use the discover_service.py it did find the right name in a really weird area, therefore i am not sure where is the issue. Maybe it's when I set up the name for my Bluetooth. I use the wrong variable or maybe bleak just does not have the ability to read the correct name?

hbldh, 2019-07-30:
During the `discover` phase, not all information available on a device is presented. To receive the information from the device one has to connect to it; something Bleak does not do during discovery and it is operation which I have no intention of adding there either.

What your iPhone does in its API has no bearing on how Windows does it in its API, so there is no guarantee that bleak using the Windows UWP Bluetooth API will return the same information.

See #73 for more information as well.

hbldh, 2019-08-02:
@DavidYauUGA @arasan90 In the version 0.5.0 which has just been release to PyPI, there is a "new" method to use for discovery:
```
from bleak.backends.dotnet.discovery import discover_by_enumeration
```
Use it the same as the `from bleak import discover`. It might see more names of your devices, but there is no guarantee.

It is the old method I had in versions <0.3.0. It uses older Windows namespaces and does not reveal Service UUIDs and manufacturer data, but I remember it as better at extracting names of devices. Give it a try.

DavidYauUGA, 2019-08-03:
> @DavidYauUGA @arasan90 In the version 0.5.0 which has just been release to PyPI, there is a "new" method to use for discovery:
>
> ```
> from bleak.backends.dotnet.discovery import [...]

Let me try that one first. Thanks anyway !!!

closed 2019-10-09 by hbldh