hbldh/bleak #98: Timeout issue with Windows

Sebwap, 2019-08-06:
* bleak version: 0.5.0
* Python version: 3.7.3
* Operating System: Windows 10

### Description

I try to connect to a BLE device with a longer timeout than default (2s)
On windows 10, if I specify a longer timeout in connection string, like "async with BleakClient(address, loop=loop, timeout=5.0) as client:" in the sensor tag example, it doesn't work.

### What I Did

while using:
`async with BleakClient(address, loop=loop, timeout=5.0) as client:`
debug show:
poll 2000.000 ms took 2000.000 ms: timeout

```

hbldh, 2019-08-06:
The `timeout` keyword only works on the `connect` method right now. Do
```
client = BleakClient(address, loop=loop)
client.connect(timeout=5.0)
# your code
client.disconnect()
```
instead of the `async with` context manager.

I had apparently missed the `with` usecase for the timeout parameter. I will remedy this in future release. Until then, try the code above.

Sebwap, 2019-08-07:
Thanks for your answer.
It's working well now, even if I still have some "Device with address xx:xx:xx:xx:xx:xx was not found".
I suspect my device (Mi Flora) to be capricious, longer timeout doesn't change almost anything.

commit 11e7adf1 (2019-10-09): Version 0.5.1

Bugfixes and improvements of client behaviour.

Merged #117
Merged #99
Merged #100
Fixes #101
Fixes #98
Fixes #104
Further fixes for #69, #83 and #86
closed 2019-10-09 by hbldh