hbldh/bleak #101: BleakError: "service is already present" when enumerating services on HUAWEI Band 2 Pro

zyv, 2019-08-18:
* bleak version: 0.5.0
* Python version: Python 3.7.4
* Operating System: macOS Mojave 10.14.6 (18G87)

### Description

Thanks for your excellent BLE library! Trying to use it to play with my wearable, but unfortunately Bleak throws an exception during service discovery.

This confuses me, because judging from the log, it actually did discover all services correctly. The device provides 4 services - information, battery, heart rate and proprietary. The two characteristics of the proprietary service are actually used to communicate over proprietary HUAWEI Link Protocol v2.

### What I Did
```python
import logging

import asyncio
from bleak import BleakClient

logging.basicConfig(level=logging.DEBUG)

async def ble_get_services(mac_addr: str, loop: asyncio.AbstractEventLoop):
    async with BleakClient(mac_addr, loop=loop) as client:
        svcs = await client.get_services()
        print("Services:", svcs)

address = "A0E49DB2-B7F1-4A65-AB2E-D75121192329"

loop = asyncio.get_event_loop()
loop.run_until_complete(ble_get_services(address, loop))
```

```
/Users/zaytsev/PycharmProjects/bluetooth/venv/bin/python /Users/zaytsev/PycharmProjects/bluetooth/bleak_demo.py
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Bluetooth powered on
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device A0E49DB2-B7F1-4A65-AB2E-D75121192329: Unknown @ RSSI: -79
A0E49DB2-B7F1-4A65-AB2E-D75121192329: HUAWEI Band 2-b6e
DEBUG:bleak.backends.corebluetooth.client:Connecting to BLE device @ A0E49DB2-B7F1-4A65-AB2E-D75121192329
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Successfully connected to device uuid A0E49DB2-B7F1-4A65-AB2E-D75121192329
WARNING:bleak.backends.corebluetooth.PeripheralDelegate:PeripheralDelegate is not compliant
DEBUG:bleak.backends.corebluetooth.client:Retrieving services...
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Services discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service 180F
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Characteristics discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A19
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A19
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service 180A
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Characteristics discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A29
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A29
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A24
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A24
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A26
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A26
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A28
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A28
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A50
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A50
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service 180D
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Characteristics discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A37
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A37
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic 2A38
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered 2A38
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service FE86
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Characteristics discovered
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic FE01
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered FE01
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic FE02
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered FE02
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic FE03
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered FE03
DEBUG:bleak.backends.corebluetooth.client:Retrieving descriptors for characteristic FE04
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Descriptor discovered FE04
DEBUG:bleak.backends.corebluetooth.client:Retrieving services...
DEBUG:bleak.backends.corebluetooth.client:Retrieving characteristics for service 180F
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Peripheral Device disconnected!
Traceback (most recent call last):
  File "/Users/zaytsev/PycharmProjects/bluetooth/bleak_demo.py", line 25, in <module>
    loop.run_until_complete(ble_get_services(address, loop))
  File "/usr/local/Cellar/python/3.7.4/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "/Users/zaytsev/PycharmProjects/bluetooth/bleak_demo.py", line 18, in ble_get_services
    svcs = await client.get_services()
  File "/Users/zaytsev/PycharmProjects/bluetooth/venv/lib/python3.7/site-packages/bleak/backends/corebluetooth/client.py", line 118, in get_services
    self.services.add_service(BleakGATTServiceCoreBluetooth(service))
  File "/Users/zaytsev/PycharmProjects/bluetooth/venv/lib/python3.7/site-packages/bleak/backends/service.py", line 101, in add_service
    "This service is already present in this BleakGATTServiceCollection!"
bleak.exc.BleakError: This service is already present in this BleakGATTServiceCollection!
```

hbldh, 2019-08-24:
Hello, thanks for reporting this. I will look into it as soon as I can. I have not encountered it myself, but it does look like something needs fixing in bleak.

zyv, 2019-08-24:
For now my workaround has be to go directly into Bleak source code and comment out duplicate checking in the abstract classes, which kinda works, but surely isn't optimal.

I'm still new to Bleak, so I'm not sure what exactly goes wrong here - does it run a second discovery somehow (apparently it's always running a discovery upon connect, so when I run it after that manually it is actually triggered for a second time), are the dictionaries supposed to be flushed between discoveries, and so on...

TomG736, 2019-09-11:
Ive just run across this too. The CoreBluetooth backend has the following code at the start of get_services()
```
if self._services != None:
    return self._services
```
self._services is never assigned in the file and the other clients have the following code instead:
```
if self._services_resolved:
    return self.services
```
swapping the code seems to work in the CoreBluetooth backend though i dont know if this is the correct fix.

hbldh, 2019-10-07:
@TomG736 Thank you for that report! I made a commit trying to solve this problem. Wil try to release it asap.

closed 2019-10-09 by hbldh

rsewell97, 2019-11-04:
Has this commit been updated to PyPI and pip? I'm still encountering this problem using pip install

hbldh, 2019-11-05:
@rsewell97 This was supposed to have been solved in v0.5.1 which is present on PyPI now. Do you encounter the exact same error? Could you please provide a traceback and then I will reopen this issue in that case.

commit 11e7adf1 (2019-10-09): Version 0.5.1

Bugfixes and improvements of client behaviour.

Merged #117
Merged #99
Merged #100
Fixes #101
Fixes #98
Fixes #104
Further fixes for #69, #83 and #86