hbldh/bleak #102: CoreBluetooth backend crash when reading protected characteristic

zyv, 2019-08-18:
* bleak version: 0.5.0
* Python version: Python 3.7.4
* Operating System: macOS Mojave 10.14.6 (18G87)

### Description

When I try reading protected characteristic, the Python process crashes, instead of throwing an exception.

### What I Did
```python
import asyncio
import logging

from bleak import BleakClient

logging.basicConfig(level=logging.DEBUG)

address = "A0E49DB2-B7F1-4A65-AB2E-D75121192329"

async def read_char(address, loop):
    async with BleakClient(address, loop=loop) as client:
        characteristic = await client.read_gatt_char("0000fe01-0000-1000-8000-00805f9b34fb")
        print(characteristic)

loop = asyncio.get_event_loop()
loop.run_until_complete(read_char(address, loop))
```

```
2019-08-18 13:17:46.582 Python[96403:3684947] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'bleak.exc.BleakError'>: Failed to read characteristic FE01: Error Domain=CBATTErrorDomain Code=2 "Reading is not permitted." UserInfo={NSLocalizedDescription=Reading is not permitted.}'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff3d2662fd __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff67933a17 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff3d27fe59 -[NSException raise] + 9
	3   _objc.cpython-37m-darwin.so         0x00000001038ca20e PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-37m-darwin.so         0x00000001038db867 method_stub + 5367
	5   _objc.cpython-37m-darwin.so         0x0000000103905560 ffi_closure_unix64_inner + 656
	6   _objc.cpython-37m-darwin.so         0x0000000103904aa6 ffi_closure_unix64 + 70
	7   CoreBluetooth                       0x00007fff3cce500e -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 115
	8   CoreBluetooth                       0x00007fff3cce0a7e -[CBPeripheral handleMsg:args:] + 297
	9   CoreBluetooth                       0x00007fff3ccdb368 -[CBCentralManager handleMsg:args:] + 198
	10  CoreBluetooth                       0x00007fff3ccd67db __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
	11  libdispatch.dylib                   0x00007fff690b35f8 _dispatch_call_block_and_release + 12
	12  libdispatch.dylib                   0x00007fff690b463d _dispatch_client_callout + 8
	13  libdispatch.dylib                   0x00007fff690ba8e0 _dispatch_lane_serial_drain + 602
	14  libdispatch.dylib                   0x00007fff690bb3c6 _dispatch_lane_invoke + 433
	15  libdispatch.dylib                   0x00007fff690bf54b _dispatch_main_queue_callback_4CF + 813
	16  CoreFoundation                      0x00007fff3d1b02d7 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	17  CoreFoundation                      0x00007fff3d1afa01 __CFRunLoopRun + 2289
	18  CoreFoundation                      0x00007fff3d1aeebe CFRunLoopRunSpecific + 455
	19  Foundation                          0x00007fff3f4137df -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 280
	20  _objc.cpython-37m-darwin.so         0x0000000103904927 ffi_call_unix64 + 79
	21  ???                                 0x00000001052c5f10 0x0 + 4381761296
)
libc++abi.dylib: terminating with uncaught exception of type NSException

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

hbldh, 2019-08-24:
Thank you for reporting this. This is bad. This one is also problematic; I am not sure how to reproduce this, having no device with protected characteristics and moreover, I am not sure I can catch that exception from bleak. 

If I made changes on the `develop`  branch, would you be available and interested in installing those versions and trying them out until we can find a solution?

zyv, 2019-08-24:
Hi @hbldh, thanks for getting back to me!

In the mean time, I have discovered that the Python process crashes on any unsupported or unauthorised operation (read from write-only | write to read-only | unauthorised reads or writes to characteristics or descriptors). Maybe this makes it easier for you to reproduce with your devices? Most cheap Chinese fitness bands have their own proprietary protocols rolled over BLE, so they often expose this kind of characteristics and/or descriptors.

Of course, I can try to help you to test your changes, it's only that I can't promise fast turn-around times... I'll be off for two weeks starting tomorrow and as I get back I'll probably only have a few spare minutes on the weekends.

bsiever, 2020-06-02:
Some of these may be resolved by PR #209 .

closed 2021-10-09 by dlech