hbldh/bleak #104: Some characteristics are missing

pierre-24, 2019-08-24:
* bleak version: 0.5.0
* Python version: 3.6.8
* Operating System: Ubuntu 18.04 (latest LTS update, so .3 or .4)

### Description

While running the [`service_explorer.py`](https://github.com/hbldh/bleak/blob/master/examples/service_explorer.py) on my device, I found that a few characteristics are missing. When I explicitely request this characteristic to be read, I get a `bleak.exc.BleakError: Characteristic with UUID xxxx could not be found`

### What I Did

I ran the [`service_explorer.py`](https://github.com/hbldh/bleak/blob/master/examples/service_explorer.py) script. The output is

```
Connected: True
[Service] f1196f50-71a4-11e6-bdf4-0800200c9a66: Unknown
	[Characteristic] f1196f57-71a4-11e6-bdf4-0800200c9a66: (write) | Name: , Value: None 
	[Characteristic] f1196f56-71a4-11e6-bdf4-0800200c9a66: (read,write) | Name: , Value: b'' 
	[Characteristic] f1196f55-71a4-11e6-bdf4-0800200c9a66: (notify) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 33) | Value: b'\x00\x00' 
	[Characteristic] f1196f54-71a4-11e6-bdf4-0800200c9a66: (read,write) | Name: , Value: b'' 
	[Characteristic] f1196f53-71a4-11e6-bdf4-0800200c9a66: (read) | Name: , Value: b'' 
	[Characteristic] f1196f52-71a4-11e6-bdf4-0800200c9a66: (read,notify) | Name: , Value: b'' 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 26) | Value: b'\x00\x00' 
	[Characteristic] f1196f51-71a4-11e6-bdf4-0800200c9a66: (read) | Name: , Value: b'' 
[Service] 0000180a-0000-1000-8000-00805f9b34fb: Device Information
	[Characteristic] 00002a26-0000-1000-8000-00805f9b34fb: (read) | Name: , Value: b'TFv3.1' 
[Service] 00001801-0000-1000-8000-00805f9b34fb: Generic Attribute Profile
	[Characteristic] 00002a05-0000-1000-8000-00805f9b34fb: (indicate) | Name: , Value: None 
		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb: (Handle: 11) | Value: b'\x02\x00' 
```
On the other hand, if I use gatttol, I get the following

```
pierre@pierre-desktop:~$ gatttool -b 98:07:2D:EE:21:0E -I
[98:07:2D:EE:21:0E][LE]> connect
Attempting to connect to 98:07:2D:EE:21:0E
Connection successful
[98:07:2D:EE:21:0E][LE]> primary
attr handle: 0x0001, end grp handle: 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x0008, end grp handle: 0x000b uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x000c, end grp handle: 0x000e uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle: 0x000f, end grp handle: 0x0014 uuid: 0000180f-0000-1000-8000-00805f9b34fb
attr handle: 0x0015, end grp handle: 0xffff uuid: f1196f50-71a4-11e6-bdf4-0800200c9a66
[98:07:2D:EE:21:0E][LE]> characteristics
handle: 0x0002, char properties: 0x0e, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb [* →  Device Name]
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb [* → Appearance]
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb [* → Peripherical Prefered Connection Parameter]
handle: 0x0009, char properties: 0x20, char value handle: 0x000a, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000d, char properties: 0x02, char value handle: 0x000e, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x0010, char properties: 0x12, char value handle: 0x0011, uuid: 00002a19-0000-1000-8000-00805f9b34fb
handle: 0x0016, char properties: 0x02, char value handle: 0x0017, uuid: f1196f51-71a4-11e6-bdf4-0800200c9a66
handle: 0x0018, char properties: 0x12, char value handle: 0x0019, uuid: f1196f52-71a4-11e6-bdf4-0800200c9a66
handle: 0x001b, char properties: 0x02, char value handle: 0x001c, uuid: f1196f53-71a4-11e6-bdf4-0800200c9a66
handle: 0x001d, char properties: 0x0a, char value handle: 0x001e, uuid: f1196f54-71a4-11e6-bdf4-0800200c9a66
handle: 0x001f, char properties: 0x10, char value handle: 0x0020, uuid: f1196f55-71a4-11e6-bdf4-0800200c9a66
handle: 0x0022, char properties: 0x0a, char value handle: 0x0023, uuid: f1196f56-71a4-11e6-bdf4-0800200c9a66
handle: 0x0024, char properties: 0x08, char value handle: 0x0025, uuid: f1196f57-71a4-11e6-bdf4-0800200c9a66
[98:07:2D:EE:21:0E][LE]> char-read-uuid 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0003 	 value: 50 69 65 72 72 65 20 2d 20 54 4d 32 
```
As you can see, the three first characteristics (marked by a star in the output, added by me) are not reported by the script. Of course, reading their value through gatttol works correctly (see last line), but as I mention, the following does not:

```
# not the smartest piece of code, but I'm just playing around ;)
import asyncio

from bleak import BleakClient

UUID_GENERIC = '0000{}-0000-1000-8000-00805f9b34fb'

CHARACTERISTIC_DEVICE_NAME = UUID_GENERIC.format('2a00')
CHARACTERISTIC_BATTERY_LEVEL = UUID_GENERIC.format('2a19')
CHARACTERISTIC_FIRMWARE_REV = UUID_GENERIC.format('2a26')

CURRENT_CHAR = CHARACTERISTIC_DEVICE_NAME

async def run(address, loop):
    async with BleakClient(address, loop=loop) as client:
        char_value = await client.read_gatt_char(CURRENT_CHAR)
        print("{} = {} ({})".format(CURRENT_CHAR, char_value.hex(), char_value.decode()))

if __name__ == '__main__':
    address = '98:07:2D:EE:21:0E'
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(address, loop))
```

Since it results in `bleak.exc.BleakError: Characteristic with UUID 00002a00-0000-1000-8000-00805f9b34fb could not be found!`. Reading `CHARACTERISTIC_BATTERY_LEVEL` does, on the other hand, works.

In itself, I'm not interested in those characteristics, but I think it worth reporting :)

hbldh, 2019-09-07:
Hello and thank you for reporting this. It is confusing, I can give you that, but it is a BlueZ backend specific thing and not entirely a bug.

The Battery level has been moved from being reported as a regular characteristic to a separate interface in the BlueZ DBus API in version 5.48. It will not show up as a characteristic in bleak due to that, but it should be handled by the current code.

The Device name was new though. I cannot find the documentation to support it, but I think the BlueZ DBus API renders that information on the device interface and I have drafted a fix for it, accessing the name where I believe it is located in the BlueZ DBus representation of the device. That cnange will be released with version 0.5.1.

The `Peripherical Prefered Connection Parameter` I have no fix for as of now. I have not device with that charactersitic, so I cannot test it. Can you give me an example value of the content of it?

Is firmware rev also problematic? I have no problems accessing it on my device by uuid?

pierre-24, 2019-09-09:
Hello. Thanks for investigating the issue :)

Firmware rev is ok, but it was appearing in `service_explorer.py`, so it should not cause any issue. 

Concerning `Peripherical Prefered Connection Parameter`, the value on my device is `50 00 a0 00 00 00 e8 03` (obtained using gatttool), but I have no idea on how to interpret this, since the documentation of this device does not mention anything on that, and I actually just report it because it shows up in the list :-°

commit 11e7adf1 (2019-10-09): Version 0.5.1

Bugfixes and improvements of client behaviour.

Merged #117
Merged #99
Merged #100
Fixes #101
Fixes #98
Fixes #104
Further fixes for #69, #83 and #86
closed 2019-10-09 by hbldh