hbldh/bleak #105: Problem with connecting to a few devices

Szymon-Gesicki, 2019-08-26:
* bleak version: 0.5.0
* Python version: 3.7.3
* Operating System: Mac os/Windows

### Description

I would like to connect with bleak to a few devices at the same moment and in one terminal. I use loop.create_task() and on windows  everything works fine, my two devices notifing perfectly. On Mac os one device also works great but if I try to connect two devices, my devices show that they are connected but they are not notifing. I debug my program and devices get in connect _to_device but print() in 'async with BleakClient() don't show.

```
from bleak import BleakClient, discover, BleakError
import asyncio

temperatureUUID = "45366e80-cf3a-11e1-9ab4-0002a5d5c51b"
ecgUUID = "46366e80-cf3a-11e1-9ab4-0002a5d5c51b"

def callback(sender, data):
    print(sender, data)

def run(addresses):
    loop = asyncio.get_event_loop()

    for address in addresses:
        loop.create_task(connect_to_device(address, loop))

    loop.run_forever()

async def connect_to_device(address, loop):

    async with BleakClient(address, loop=loop) as client:

        print("connect to ", address)

        await client.start_notify(ecgUUID, callback)

        while True:
            await asyncio.sleep(1, loop=loop)

if __name__ == "__main__":
    run(["9923F33A-0648-4FC4-A062-47C872A4C27D","4CFEB2B0-02E7-46E4-B80D-0B1F5D4AD7CC"])
```

hbldh, 2019-09-07:
Can you try to connect to two devices in macOS using two different terminals and see if that works. If so, then there is a limitation in the bleak macOS backend that needs to be documented.

See this for similar problem: https://stackoverflow.com/questions/34787976/how-can-i-implement-core-bluetooth-functionality-for-multiple-devices

Szymon-Gesicki, 2020-05-14:
It took me a while 😀 I did what you said. Bleak connects correctly to two devices at two different terminals. Unfortunately, just like in my example, it stops before
async with BleakClient(address, loop=loop) as client:. 
On Windows everything works perfect.

hbldh, 2020-05-15:
Ok. I will add a note about this in the documentation for the time being. This issue is beyond the scope of my available time to address.

hbldh, 2020-06-30:
This is solved in the last `develop` branch commit and will be available on PyPI in version 0.7.0.

The following code should now work:
```python
from bleak import BleakClient
import asyncio

notify_uuid = "0000{0:x}-0000-1000-8000-00805f9b34fb".format(0xffe1)

def callback(sender, data):
    print(sender, data)

def run(addresses):
    loop = asyncio.get_event_loop()

    tasks = asyncio.gather(
        *(connect_to_device(address, loop) for address in addresses)
    )

    loop.run_until_complete(tasks)

async def connect_to_device(address, loop):
    print("starting", address, "loop")
    async with BleakClient(address, loop=loop, timeout=10.0) as client:

        print("connect to", address)
        try:
            await client.start_notify(notify_uuid, callback)
            await asyncio.sleep(10.0, loop=loop)
            await client.stop_notify(notify_uuid)
        except Exception as e:
            print(e)

    print("disconnect from", address)

if __name__ == "__main__":
    run(["B9EA5233-37EF-4DD6-87A8-2A875E821C46", "F0CBEBD3-299B-4139-A9FC-44618C720157" ])
```

closed 2020-06-30 by hbldh

sd1026, 2021-07-16:
I have been able to connect to my 5 BLE modules, and I am receiving notifications from all of them. But I have a doubt that when I pass data through the callback function, it takes (sender, data) as parameters. Now, here I am confused as to what is this "sender" here. When i debug the code, it always prints 13 as sender.

So, is it the case that multiple ble modules are connected but, the notifications are coming from only one of them.
Or, is it something that the code already has.

commit ad2f8eb4 (2020-06-30): Changelog entry about fixes in #227.

Fixes #105
Fixes #111
Fixes #211
Fixes #206