hbldh/bleak #107: Notification handler passed to start_notify cannot raise uncaught exception without leading to a cryptic crash

Arrowana, 2019-09-03:
* bleak version: 0.5.0
* Python version:  3.7.4
* Operating System: MacOS Mojave

### Description

Notification handler passed to start_notify cannot raise uncaught exception without leading to a cryptic crash

### What I Did

Using the enable_notification.py sample, I tried to add some code to the notification handler in order to parse the notified message. However, my code contained a bug which caused the code to raise an exception. I simplified the problem and realised that any exception in the notification handler will result in a terrible crash.

```
def p():
    print('Yo')
    raise ArithmeticError

def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    print("{0}: {1}".format(sender, data))
    #csc_delegate.handleNotification(1, data)
    p()
```

which lead to

```
Yo
2019-09-03 19:57:12.341 Python[54571:306992] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'ArithmeticError'>: '
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff2cedacfd __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff57580a17 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff2cef485d -[NSException raise] + 9
	3   _objc.cpython-37m-darwin.so         0x000000010763b20e PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-37m-darwin.so         0x000000010764c867 method_stub + 5367
	5   _objc.cpython-37m-darwin.so         0x0000000107676560 ffi_closure_unix64_inner + 656
	6   _objc.cpython-37m-darwin.so         0x0000000107675aa6 ffi_closure_unix64 + 70
	7   CoreBluetooth                       0x00007fff2c963022 -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 115
	8   CoreBluetooth                       0x00007fff2c95ea92 -[CBPeripheral handleMsg:args:] + 297
	9   CoreBluetooth                       0x00007fff2c95937c -[CBCentralManager handleMsg:args:] + 198
	10  CoreBluetooth                       0x00007fff2c9547ef __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
	11  libdispatch.dylib                   0x00007fff58d005f8 _dispatch_call_block_and_release + 12
	12  libdispatch.dylib                   0x00007fff58d0163d _dispatch_client_callout + 8
	13  libdispatch.dylib                   0x00007fff58d078e0 _dispatch_lane_serial_drain + 602
	14  libdispatch.dylib                   0x00007fff58d083c6 _dispatch_lane_invoke + 433
	15  libdispatch.dylib                   0x00007fff58d0c54b _dispatch_main_queue_callback_4CF + 813
	16  CoreFoundation                      0x00007fff2ce24cd7 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	17  CoreFoundation                      0x00007fff2ce24401 __CFRunLoopRun + 2289
	18  CoreFoundation                      0x00007fff2ce238be CFRunLoopRunSpecific + 455
	19  Foundation                          0x00007fff2f0878ef -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 280
	20  _objc.cpython-37m-darwin.so         0x0000000107675927 ffi_call_unix64 + 79
	21  ???                                 0x000000010aa44610 0x0 + 4473505296
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

Shouldn't it raise the python exception instead and keep going?

zyv, 2019-09-15:
I think that this is a duplicate of #102 - basically any unhandled Python exception inside the loop causes a segfault.

hbldh, 2019-10-07:
This is bad and should be improved. I have very little capacity to do any kind of development on this project, and even less to do debugging on macOS, the platform I use the least frequent. I have no estimate on when this could be addressed. 
I'm sorry for this, but cannot do anything about it right now.

Arrowana, 2019-10-08:
Well, I can have a look into the issue. However, what should be the expected behavior?
After an exception should the callback be removed? Or should we allow a raising callback to be called again?

Arrowana, 2019-10-08:
https://pyobjc.readthedocs.io/en/latest/api/module-objc.html According to this documentation, the whole purpose of objc.options.verbose = True is to log when a python exception is converted to objc.
As a result, setting it to True, leads to having the python exception:
```
Character Notify Update
2A5B: b'\x01\x04\x00\x00\x00\xa0\x02'
2019-10-08 18:33:48.856 Python[2613:200631] PyObjC: Converting exception to Objective-C:
Traceback (most recent call last):
  File "/Users/yenkooy/PIERRE/bleak/bleak/backends/corebluetooth/PeripheralDelegate.py", line 253, in peripheral_didUpdateValueForCharacteristic_error_
    self._characteristic_notify_callbacks[cUUID](cUUID, characteristic.value())
  File "bleak_csc.py", line 14, in notification_handler
    raise RuntimeError('Hello')
RuntimeError: Hello
2019-10-08 18:33:48.857 Python[2613:200631] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'RuntimeError'>: Hello'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff47a5a2fd __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x00007fff7212ba17 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff47a73e59 -[NSException raise] + 9
	3   _objc.cpython-37m-darwin.so         0x000000010739520e PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-37m-darwin.so         0x00000001073a6867 method_stub + 5367
	5   _objc.cpython-37m-darwin.so         0x00000001073d0560 ffi_closure_unix64_inner + 656
	6   _objc.cpython-37m-darwin.so         0x00000001073cfaa6 ffi_closure_unix64 + 70
	7   CoreBluetooth                       0x00007fff474d900e -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 115
	8   CoreBluetooth                       0x00007fff474d4a7e -[CBPeripheral handleMsg:args:] + 297
	9   CoreBluetooth                       0x00007fff474cf368 -[CBCentralManager handleMsg:args:] + 198
	10  CoreBluetooth                       0x00007fff474ca7db __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
	11  libdispatch.dylib                   0x00007fff738ab5f8 _dispatch_call_block_and_release + 12
	12  libdispatch.dylib                   0x00007fff738ac63d _dispatch_client_callout + 8
	13  libdispatch.dylib                   0x00007fff738b28e0 _dispatch_lane_serial_drain + 602
	14  libdispatch.dylib                   0x00007fff738b33c6 _dispatch_lane_invoke + 433
	15  libdispatch.dylib                   0x00007fff738b754b _dispatch_main_queue_callback_4CF + 813
	16  CoreFoundation                      0x00007fff479a42d7 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	17  CoreFoundation                      0x00007fff479a3a01 __CFRunLoopRun + 2289
	18  CoreFoundation                      0x00007fff479a2ebe CFRunLoopRunSpecific + 455
	19  Foundation                          0x00007fff49c0732f -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 280
	20  _objc.cpython-37m-darwin.so         0x00000001073cf927 ffi_call_unix64 + 79
	21  ???                                 0x000000010a7922b0 0x0 + 4470678192
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6
```

With the notification_handler being
```
def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    print("{0}: {1}".format(sender, data))
    raise RuntimeError('Hello')
```

Is it good enough, at least it works for me. I get to know what causes the crash?
#117

hbldh, 2019-10-08:
The expected behaviour IMO is that whenever an exception is raised in Core Bluetooth it should be caught and transformed (at least the message) into a BleakError and raised as a python exception so that the user can catch it.

Can you catch that exception in your example with the `objc.options.verbose = True`?

A solution of reasonably low complexity is to instruct users to always wrap their notification handler code in macOS in a try-except block:
```
def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    try:
        print("{0}: {1}".format(sender, data))
        raise RuntimeError('Hello')
    except Exception as e:
        logger.info("Error in notification_handler: " + e.message)
```
That way you get information about error, but the notification error is not sent back into Core Bluetooth and raised there as a NSException.

hbldh, 2019-10-09:
This is addressed in #117 - improved but not fixed - and this issue will thus be closed.

closed 2019-10-09 by hbldh