hbldh/bleak #108: BleakClient does not correctly deregister callback function upon disconnect/delete

Mithrandir2k18, 2019-09-04:
* bleak version: 0.4.3
* Python version: 3.5
* Operating System: Raspia Streatch

### Description
I am maintaining multiple BLEClients to multiple devices that get in and out of range regularly. If I have a connection to a device I discover, I do nothing, else I create a new BleakClient object.

On disconnect I call .disconnect() and delete the Object.

When the device get back in range, I create a new BLEClient and register a notify callback. But then for each piece of data the callback now gets called twice. The callback from the disconnected and deleted client is still active and reactivates itself, which is undocumented behaviour.

Solution: Correctly deregister the callback upon calling BleakClient.disconnect() or del BleakClient, or document this auto-reconnect behaviour.

However if disconnecting without terminating the process is a desired feature, this issue has to be fixed.

```python
 while True:
        for device_mac, _ in devices:  # set of mac,name tuples which I want to connect to if in range
            if device_mac not in ble_clients.keys():
                ble_clients.update(
                    {device_mac: bleak.BleakClient(address=device_mac,
                                                   loop=loop)})
                try:
                    await ble_clients.get(device_mac).connect(loop=loop)
                    await on_connect(ble_clients.get(device_mac))
                except Exception:
                    logging.exception(
                        "An error occurred while connecting to BLE device: "
                        + device_mac)
            else:
                try:
                    disconnected = await ble_clients.get(device_mac).is_connected() is False
                    if disconnected:
                        await ble_clients.get(device_mac).disconnect()
                        raise RuntimeError(
                            "Disconnected BLE device with MAC: " + device_mac)
                except Exception as e:
                    logging.exception(
                        "Device with MAC disconnected: " + str(e))
                    del ble_clients[device_mac]
```

hbldh, 2019-09-04:
This should be fixed in version 0.5.0 of Bleak, see #89. Upgrade and see if it is a problem there still.

Mithrandir2k18, 2019-09-04:
Cannot confirm. Behaviour is really weird.

Right now if on version 0.4.3 I delete the entire else block in the code above, the results are as desired, however I won't have any objects of my clients anymore, meaning terminating the connection/callbacks is only possible by ending the process.

In 0.5.0 if I keep the else block where the deletion of the client happens, I again get duplicates as described above. If remove the code of the else block again, the client that is kept alive forever never reconnects after the device gets back in range/is reactivated.

This means that the desired behaviour is unachievable in 0.5.0. And in 0.4.3 it is, but only in a dirty fashion.

Is my approach flawed? How would you handle multiple clients that come and go regularly?

hbldh, 2019-09-04:
In version 0.5.0 there is a callback for the disconnect event; check out #86 and the method and 
 documentation [here](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/client.py#L57-L69). Try that out, to make sure that subscriptions are deleted in the BlueZ DBus API on disconnect.

Otherwise maybe @JoeHut could help you along better than I can, since I haven't used bleak in that way myself.

Mithrandir2k18, 2019-09-05:
But isn't that also achieved by my call to the `.disconnect()` method? What should I register as a disconnect_callback? A direct dbus/bluez api call? Also why isn't terminating all subscriptions handled in a `BleakClient.__del__()` method?

Mithrandir2k18, 2019-09-05:
I just tried to implement what I want using the disconnected callbacks. Upon executing my function I get a `BleakClientBlueZDBus object has no attribute set_disconnected_callback` and `_disconnected_callback` also doesn't exist.

Printing `bleak.BleakClient(real_mac_addr).__dict__.keys()` results in:
`dict_keys(['address', '_device_path', 'services', '_notification_callbacks', '_bluez_version', '_services_resolved', '_rules', '_bus', '_char_path_to_uuid', 'device', 'loop']`

Howver printing `bleak.BleakClient("").__dict__.keys()` results in `dict_keys(['_disconnected_callback', '_device_path', '_rules', 'services', 'loop', '_char_path_to_uuid', '_bus', 'address', '_subscriptions', 'device', '_notification_callbacks', '_bluez_version', '_services_resolved'])`

Something goes wrong when creating the connection and the returned object doesn't get/keep the disconnected callback related attributes.

hbldh, 2019-09-07:
The changes implemented in 749f27c just now could solve your problems. Either you install the `develop` branch version or wait until I release it in version 0.5.1. Upon disconnect received from BlueZ, all notifications from that BleakClient are removed. There is a new [example](https://github.com/hbldh/bleak/blob/develop/examples/disconnect_callback.py) in the `develop` branch showing a more generic example on how to use the method. Even without a disconnect callback, all notifications are cleared upon getting a notification from BlueZ that the device is disconnected.

The reason you do not see the method must be that you are not using version 0.5.0. I confirmed that the source files of 0.5.0 on PyPI contains clients with that very method.

I never implement `__del__` methods. IMO you should avoid calling them yourself. [They are probably not what you are looking for](https://stackoverflow.com/a/1481512). Use `ble_clients.pop(device_mac)` instead to just remove it from the dict and move on.

Mithrandir2k18, 2019-09-16:
> Use ble_clients.pop(device_mac) instead to just remove it from the dict and move on.
Since this just got referenced, I'll clarify that `del` and `.pop()` both delete the value from the dict as desired, with the only difference that `.pop()` preserves the object and returns it, where as `del` doesn't return anything and deletes the object [as intended](https://docs.python.org/3/library/stdtypes.html#mapping-types-dict).

hbldh, 2019-10-08:
I have no desire to argue `__del__` methods. But yes, `__del__` on a dict does just that, as your link says. My comment on `__del__` was about your desire to "terminating all subscriptions handled in a BleakClient.__del__() method". For that I prefer to use a method defined by me. My response above was ambiguous I will give you that, going on about `ble_clients.pop` instead.

I will try to replicate this from a 0.5.0 install. It shoudl not happen, but if you say it does I will have to check it out.

hbldh, 2019-11-01:
I have been analyzing this issue now and I have done what I can with it.

You should preferably use `async with` context managers. If you can't, make sure you call `disconnect` (or `_cleanup`) on each client that you dispose of/reconnect with. The `__del__` method cannot be used to clean up DBUS since it is not asynchronous, as far as I can tell.

I have further modified the disconnect callback solution and it is working fine for me (See the `disconnect_callback`). (Changes affecting #86) These improvements will be present in version 0.5.2. I do not have multiple devices to test it with though so I do not know how it handles there. You will have to navigate those waters yourself.

PR #164 (2020-03-09): Release/v0.6.0
* New Scanner object to allow for async device scanning.
* Updated ``txdbus`` requirement to version 1.1.1 (Merged #122)
* Implemented ``write_gatt_descriptor`` for Bluez backend.
* Large change in Bluez backend handling of Twisted reactors. Fixes #143
* Modified ``set_disconnect_callback`` to actually call the callback as a callback. Fixes #108.
* Added another required parameter to disconnect callbacks.
* Added Discovery filter option in BlueZ backend (Merged #124)
* Merge #138: comments about Bluez version check.
* Improved scanning data for macOS backend. Merge #126.
* Merge [... cut]
closed 2020-03-09 by hbldh