hbldh/bleak #109: No module named 'Windows' / Bleak can not connect on Windows

sonus89, 2019-09-09:
* bleak version: 0.5.0
* Python version: 3.7.4
* Operating System: Windows 10 (Build 18362)

### Description

I receive an error message:

Traceback (most recent call last):
  File "...\blue.py", line 2, in <module>
    from bleak import BleakClient
  File "...\venv\lib\site-packages\bleak\__init__.py", line 68, in <module>
    from bleak.backends.dotnet.discovery import discover  # noqa
  File "...\venv\lib\site-packages\bleak\backends\dotnet\discovery.py", line 17, in <module>
    from Windows.Devices.Bluetooth.Advertisement import BluetoothLEAdvertisementWatcher

**ModuleNotFoundError: No module named 'Windows'**
Process finished with exit code 1

### What I Did

I executed the bleak connect example from here:
https://github.com/hbldh/bleak

hbldh, 2019-11-01:
Sorry for a very late reply... Are you still interested in a solution to this?

sonus89, 2019-11-11:
Hi,

Do you know how can I check with bleak if the bluetooth module is switched
off?
I would like to inform the user in my program to switch it on.

Thanks,
Máté

Henrik Blidh <notifications@github.com> ezt írta (időpont: 2019. nov. 1.,
P, 20:24):

hbldh, 2019-11-12:
No. It is not part of bleak right now.

See if this can help: https://stackoverflow.com/a/48794678

closed 2019-11-12 by hbldh

positron96, 2020-01-02:
Hi, I'm also getting this error:
```
C:\Users\positron\AppData\Roaming\Python\Python37\Scripts>C:\Users\positron\AppData\Roaming\Python\Python37\Scripts\bleak-lescan.exe   Traceback (most recent call last):                                                                                                       File "c:\program files\python37\lib\runpy.py", line 193, in _run_module_as_main                                                          "__main__", mod_spec)                                                                                                                File "c:\program files\python37\lib\runpy.py", line 85, in _run_code                                                                     exec(code, run_globals)                                                                                                              File "C:\Users\positron\AppData\Roaming\Python\Python37\Scripts\bleak-lescan.exe\__main__.py", line 5, in <module>                     File "C:\Users\positron\AppData\Roaming\Python\Python37\site-packages\bleak\__init__.py", line 68, in <module>                           from bleak.backends.dotnet.discovery import discover  # noqa                                                                         File "C:\Users\positron\AppData\Roaming\Python\Python37\site-packages\bleak\backends\dotnet\discovery.py", line 17, in <module>          from Windows.Devices.Bluetooth.Advertisement import BluetoothLEAdvertisementWatcher                                                ModuleNotFoundError: No module named 'Windows'                                                                                          
```

I've tried to launch bleak-lescan.exe. 
How do I fix this?

System is Windows 10 latest,
python 3.7.6

bluebrick-Adafruit-BluefruitLE==0.9.13
bricknil==0.9.3
bricknil-bleak==0.3.1
curio==0.9
future==0.18.2
pythonnet==2.4.0
PyYAML==5.2

positron96, 2020-01-02:
Also reported here: https://github.com/virantha/bricknil/issues/3