hbldh/bleak #111: Bleak CoreBluetooth backend is leaking tasks?

zyv, 2019-09-15:
* bleak version: 0.5.0
* Python version: Python 3.7.4
* Operating System: macOS Mojave 10.14.6 (18G87)

### Description

If I close the event loop after using Bleak and disconnecting from the device, I get an error from asyncio saying that it destroyed a pending task:

```
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Peripheral Device disconnected!
ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<Application._handle_nsrunloop() running at /Users/zaytsev/PycharmProjects/huawei-lpv2/venv/lib/python3.7/site-packages/bleak/backends/corebluetooth/__init__.py:45>>
```

### What I Did

```
    event_loop = asyncio.get_event_loop()

    try:
        event_loop.run_until_complete(run(...))
    finally:
        event_loop.close()
```

hbldh, 2019-10-07:
Try adding a
```python
from bleak.backends.corebluetooth.client import cbapp
cbapp.ns_run_loop_done = True
await event_loop.sleep(0.1)
```
somewhere in the end of your `run` method.

That might remove the warning.

hbldh, 2020-06-02:
Have not heard this from anyone else since. Will close for now.

closed 2020-06-02 by hbldh

dlech, 2020-06-18:
This only halfway fixes the problem `_central_manager_delegate_ready` can still be pending.

commit ad2f8eb4 (2020-06-30): Changelog entry about fixes in #227.

Fixes #105
Fixes #111
Fixes #211
Fixes #206
PR #232 (2020-06-30): Release/v0.7.0
# Version 0.7.0 (2020-06-30)

## Added

* Better feedback of communication errors to user in .NET backend and implementing error details proposed in #174.
* Two devices example file to use for e.g. debugging.
* Detection/discovery callbacks in Core Bluetooth backend ``Scanner`` implemented.
* Characteristic handle printout in ``service_explorer.py``.
* Added scanning filters to .NET backend's ``discover`` method.

## Changed

* Replace ``NSRunLoop`` with dispatch queue in Core Bluetooth backend. This causes callbacks to be dispatched on a background thread instead of on the main [... cut]