hbldh/bleak #112: write_gatt_descriptor raising "NotImplementedError"

wsidl, 2019-09-15:
* bleak version: 0.5.0
* Python version: 3.7.4
* Operating System: Linux (Fedora 30, kernel 5.2.11)

### Description
BLE Device requires writing commands to get a value from it's internals, but write_gatt_descriptor always raises "NotImplementedError". Looking into the extended device object, the method calls the super of the method without following up without any instructions.

```python
    async def write_gatt_descriptor(self, handle: int, data: bytearray) -> None:

        """Perform a write operation on the specified GATT descriptor.

        Args:
            handle (int): The handle of the descriptor to read from.
            data (bytes or bytearray): The data to send.

        """
        return await super().write_gatt_descriptor(handle, data)

    async def start_notify(
...
```
Is this method not supported yet? What is needed to get this to work?

### What I Did
The code is a direct pull from "enable_notifications" example code, except in the run method, the write method was added to have something for the notifier to catch.

```python
    async with BleakClient(address, loop=loop) as client:
        x = await client.is_connected()
        await client.start_notify(CHARACTERISTIC_UUID, notification_handler)
        logger.info("Writing command")
        await client.write_gatt_descriptor(0x12, bytearray('my command\r', 'utf8'))
        logger.info("Waiting for notification")
        await asyncio.sleep(5.0)
        logger.info("Stopping Notifier")
        await client.stop_notify(CHARACTERISTIC_UUID)
```

This returns with NotImplementedError() on "write_gatt_descriptor".

```
  File "lib64/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 413, in write_gatt_descriptor
    return await super().write_gatt_descriptor(handle, data)
  File "python3.7/site-packages/bleak/backends/client.py", line 166, in write_gatt_descriptor
    raise NotImplementedError()
```

wsidl, 2019-09-15:
Should add that the values were tested using gatttool:
```
[FF:FF:FF:FF:FF:FF][LE]> char-write-cmd 0x0012 0123456789abcdef0d
Notification handle = 0x0012 value: 01 23 45 67 89 ab cd ef 0d 
```

wsidl, 2019-09-16:
Also tried extending BleakClient to implement the method based on what I saw from the other methods, but can't seem to get any further

```python

class NewClient(BleakClient):
    async def write_gatt_descriptor(self, handle: int, data: bytearray) -> None:
        descriptor = self.services.get_descriptor(handle)
        await self._bus.callRemote(
            descriptor.path,
            'WriteValue',
            interface=defs.GATT_DESCRIPTOR_INTERFACE,
            destination=defs.BLUEZ_SERVICE,
            signature='a{sv}',
            body=[data, {'type': 'command'}],
            returnSignature='',
        ).asFuture(self.loop)

        bluezlogger.debug(
            "Write Descriptor {0} | {1}: {2}".format(
                handle, descriptor.path, data
            )
        )
```
This usually just results in "TypeError: zip argument #⁠2 must support iteration"

hbldh, 2019-10-07:
This is an oversight. I seems to have forgotten to implement this, even though the method should exist (https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt#n308).

I think you are on the right track though. In the link above, the method does not take any `type`-parameter and I think the signature should be `"aya{sv}"` instead.

Try that out and if it works, feel free to send a PR!

wsidl, 2019-10-13:
Thanks, I've implemented that change and something happened (no errors), but after implementing it found that the device I'm using uses a write characteristic. Until I can find a device that provides a write descriptor endpoint, I'll keep this on the side.

hbldh, 2019-11-01:
I will close this issue in the meantime. If you do find a device that can handle this, please write this in this issue.

I have integrated the code you provided and will release it in version 0.5.2. Thank you for the help!

commit 1da0b6c1 (2019-11-01): Implemented (untested) write_gatt_char for BlueZ backend.

Hopefully fixes #112
commit bdd90b64 (2020-03-09): Implemented (untested) write_gatt_char for BlueZ backend.

Hopefully fixes #112
closed 2019-11-01 by hbldh