hbldh/bleak #114: write with/without response is not valid

fanshuaiqi, 2019-09-20:
* bleak version:0.5.0
* Python version:3.7
* Operating System:mac os 10.14.6

### Description
base on my understanding, write_gatt_char can set write with/without response, but the arg of this function is not passed to next level call.It uses self.peripheral.writeValue_forCharacteristic_type_(value, characteristic, 0) to write char eventually.

hbldh, 2019-10-08:
Is the lack of propagation a problem for you or is it a general comment on the proper way of doing this? I agree that it is a lack of coherence, but I have never experienced myself that the lack of using a write instead of a write-with-response or vice versa is that much of an obstacle?

5frank, 2020-01-06:
The fix raises the following error for me. 
```
objc.BadPrototypeError: Objective-C expects 3 arguments, Python argument has 4 arguments for <unbound selector writeCharacteristic:value: of PeripheralDelegate at 0x1036ccce0>
```

dlech, 2020-01-06:
Made some (untested) suggestions on possibly how to fix: https://github.com/hbldh/bleak/commit/75f2ff9f3a1ff7775a062072138b33d588d418aa

5frank, 2020-01-06:
the underscores seems to be of importance for pyobjc. should match number of args. (I'm new to this).
The following seems to fix it but I'm on a messy branch.
```diff
diff --git a/bleak/backends/corebluetooth/PeripheralDelegate.py b/bleak/backends/corebluetooth/PeripheralDelegate.py
index 9fa73b3..0e2d0c5 100644
--- a/bleak/backends/corebluetooth/PeripheralDelegate.py
+++ b/bleak/backends/corebluetooth/PeripheralDelegate.py
@@ -145,10 +145,10 @@ class PeripheralDelegate(NSObject):
 
         return descriptor.value()
 
-    async def writeCharacteristic_value_(
+
+    async def writeCharacteristic_value_type_(
         self, characteristic: CBCharacteristic, value: NSData, response: int
     ) -> bool:
-
         cUUID = characteristic.UUID().UUIDString()
         self._characteristic_write_log[cUUID] = False
 
diff --git a/bleak/backends/corebluetooth/client.py b/bleak/backends/corebluetooth/client.py
index 8ac39ba..63e8b37 100644
--- a/bleak/backends/corebluetooth/client.py
+++ b/bleak/backends/corebluetooth/client.py
@@ -220,7 +220,7 @@ class BleakClientCoreBluetooth(BaseBleakClient):
             raise BleakError("Characteristic {} was not found!".format(_uuid))
 
         value = NSData.alloc().initWithBytes_length_(data, len(data))
-        success = await cbapp.central_manager_delegate.connected_peripheral_delegate.writeCharacteristic_value_(
+        success = await cbapp.central_manager_delegate.connected_peripheral_delegate.writeCharacteristic_value_type_(
             characteristic.obj, value, 1 if response else 0
         )
         if success:
```

5frank, 2020-01-06:
"since each colon in an Objective-C selector is a placeholder for an argument, the number of underscores in the PyObjC-ified method name is the number of arguments that should be given." https://pyobjc.readthedocs.io/en/latest/core/intro.html

hbldh, 2020-01-06:
Crap. That was a clumsy fix by me, forgetting that naming stuff...

 I will try to straighten it out tomorrow...

hbldh, 2020-01-07:
Committed a fix that is now tested for real as well. Sorry and thank you both for guidance and patience.

commit 75f2ff9f (2020-01-05): Fixes #114.
commit f2aa8d93 (2020-01-07): Fixes #114 and corrected bug entered by 75f2ff9
commit e2e9f7fe (2020-03-09): Fixes #114.
PR #164 (2020-03-09): Release/v0.6.0
* New Scanner object to allow for async device scanning.
* Updated ``txdbus`` requirement to version 1.1.1 (Merged #122)
* Implemented ``write_gatt_descriptor`` for Bluez backend.
* Large change in Bluez backend handling of Twisted reactors. Fixes #143
* Modified ``set_disconnect_callback`` to actually call the callback as a callback. Fixes #108.
* Added another required parameter to disconnect callbacks.
* Added Discovery filter option in BlueZ backend (Merged #124)
* Merge #138: comments about Bluez version check.
* Improved scanning data for macOS backend. Merge #126.
* Merge [... cut]
closed 2020-03-09 by hbldh