hbldh/bleak #116: Hanging on "async with BleakClient..."

GreatScott, 2019-10-03:
* bleak version: 0.5.0-develop/branch
* Python version: 3.7
* Operating System: macOS 10.14.6 (on early 2015 Macbook Air)

### Description
In running the examples on MacOS, everything hangs after/within `async with BleakClient(address, loop=loop) as client:`  (_within_ `run`) i.e. the subsequent typical example `x = await client.is_connected` or any other calls do not appear to run. Despite this, the OS does indicate that the BLE device _is_ connected indefinitely until the script is stopped. Everything appears to be working fine on Windows.

hbldh, 2019-10-08:
I will try to replicate this and see if I can verify it, but my trials with a Mac Mini with an even older macOS version did not exibit this behaviour. Did this fail with all examples? Or just some? Which in that case?

GreatScott, 2019-10-08:
Thanks for the response! I'll run through all the examples in a bit and report back.

alamysm, 2019-12-24:
I'm having the same issue, I think. I'm at the top of the develop branch ("0.5.2a1 (2019-11-10)"?) on Mac OS 10.14.6 on a 2019 MBP 13. I've tested with Python 3.7.6 and 3.8.1 with the same results. The discover example appears to work as expected, but get_services and service_explorer exhibit the described behavior. Here are my logs for service_explorer:

`DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 3DBBDC7D-4F6E-401F-8154-7B8ACE865CAE: Unknown @ RSSI: -97
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device F44CAE74-564E-4F68-9FA4-8BA1D78EFFFB: Unknown @ RSSI: -84
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Discovered device 7F242CAE-B671-4530-8F2D-6F5D0B4874E6: Unknown @ RSSI: -83
DEBUG:bleak.backends.corebluetooth.client:Connecting to BLE device @ F44CAE74-564E-4F68-9FA4-8BA1D78EFFFB
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Successfully connected to device uuid F44CAE74-564E-4F68-9FA4-8BA1D78EFFFB
WARNING:bleak.backends.corebluetooth.PeripheralDelegate:PeripheralDelegate is not compliant
DEBUG:bleak.backends.corebluetooth.client:Retrieving services...
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Successfully connected to device uuid F44CAE74-564E-4F68-9FA4-8BA1D78EFFFB
WARNING:bleak.backends.corebluetooth.PeripheralDelegate:PeripheralDelegate is not compliant
DEBUG:bleak.backends.corebluetooth.PeripheralDelegate:Services discovered
DEBUG:bleak.backends.corebluetooth.CentralManagerDelegate:Peripheral Device disconnected!`

(That last 'disconnected' line happens a minute after the previous line probably because my BLE peripheral device has a 60-second inactivity timeout.)

My Python experience is relatively limited (especially Python 3.x), but let me know if there's anything I can do to help with this.

adammark, 2020-02-18:
I've been playing with the develop branch and I saw something similar happen when connecting to certain devices with OS X. The best I can tell is that `centralManager_didConnectPeripheral_` is being fired twice, creating two peripheral delegates. The first delegate proceeds with discovering services and characteristics, but by this point the central manager has a handle to a different delegate, which I suppose is either doing nothing or is otherwise blocked. Therefore  `client.connect()` never resolves.

I can't say why the callback is being fired twice for some devices and not others, but I was able to work around the problem with a one line check here:

``` python
def centralManager_didConnectPeripheral_(self, central, peripheral):
    logger.debug(
        "Successfully connected to device uuid {}".format(
            peripheral.identifier().UUIDString()
        )
    )
    if not self.connected_peripheral_delegate:
        peripheralDelegate = PeripheralDelegate.alloc().initWithPeripheral_(peripheral)
        self.connected_peripheral_delegate = peripheralDelegate
        self._connection_state = CMDConnectionState.CONNECTED
```

ShaharHD, 2020-04-03:
Hi, having the exact same issue. I assume this is still not resolved?

hbldh, 2020-04-06:
@adammark @ShaharHD Which versions of Bleak did you run?

adammark, 2020-04-06:
I was using the develop branch, mid February.

ShaharHD, 2020-04-07:
I pulled in whatever pip pulled:

 `cat __version__.py`

```python
# -*- coding: utf-8 -*-

__version__ = "0.6.1"
```

I have generated a diff file based on @adammark comment, and used it to patch the version after `pip install`

```diff
@@ -202,9 +202,10 @@
                 peripheral.identifier().UUIDString()
             )
         )
-        peripheralDelegate = PeripheralDelegate.alloc().initWithPeripheral_(peripheral)
-        self.connected_peripheral_delegate = peripheralDelegate
-        self._connection_state = CMDConnectionState.CONNECTED
+        if not self.connected_peripheral_delegate:
+            peripheralDelegate = PeripheralDelegate.alloc().initWithPeripheral_(peripheral)
+            self.connected_peripheral_delegate = peripheralDelegate
+            self._connection_state = CMDConnectionState.CONNECTED
 
     def centralManager_didFailToConnectPeripheral_error_(
         self, centralManager: CBCentralManager, peripheral: CBPeripheral, error: NSError
```

This solved the issue for me - but now has to be patched part of each `pip install`

corvis, 2020-04-14:
I have the same issue and could confirm the fix from @ShaharHD solves it. 
Any chances this will be added to upstream and included in the next release, @hbldh?

hbldh, 2020-04-21:
Make a pull request to `develop`, solving this issue and I will merge it and schedule it for the 0.6.2 release.

dderiso, 2020-06-06:
@ShaharHD's fix worked for me too.

ShaharHD, 2020-06-07:
> Make a pull request to `develop`, solving this issue and I will merge it and schedule it for the 0.6.2 release.

Created #213

hbldh, 2020-06-30:
I have merged #227 into develop and will make a version 0.7.0 release later today with those changes as well as a lot of others. If the "fire twice" problem is still left after release of version 0.7.0, I will include the changes in #213 in a version 0.7.X release as soon as someone confirms that it is still present.

sn-uaudio, 2020-07-02:
>  I will include the changes in #213 in a version 0.7.X release as soon as someone confirms that it is still present.

Hello, I can confirm that the issue is still present for me on macOS 10.14.6. I updated to bleak 0.7.0 and was still having this problem. I added this line as a workaround and it seems to have solved the issue for me:

`    @objc.python_method
    def did_connect_peripheral(self, central, peripheral):
        logger.debug(
            "Successfully connected to device uuid {}".format(
                peripheral.identifier().UUIDString()
            )
        )
        **if not self.connected_peripheral_delegate:**
            peripheralDelegate = PeripheralDelegate.alloc().initWithPeripheral_(peripheral)
            self.connected_peripheral_delegate = peripheralDelegate
            self._connection_state = CMDConnectionState.CONNECTED
`

hbldh, 2020-07-02:
Sigh. Ok. Could someone please provide me with a minimal example where this happens, so I can understand why this issue will not be solved by previous efforts?

If it fires twice it feels inadequate to just skip that by using an if-statement and move on. I do not want to treat symptoms but rather find the root cause, regardless if #213 solves it or not.

ShaharHD, 2020-07-03:
> Sigh. Ok. Could someone please provide me with a minimal example where this happens, so I can understand why this issue will not be solved by previous efforts?
> 

Do you have any BLE device (Nordic, Silicon Labs, etc) I could give you a sample code for Mac and firmware for the device to test with?

hbldh, 2020-07-09:
I have no devices that I want to program with specific firmware to be able to reproduce this. If it does not happen with every device, but with non-standard firmware on certain devices, I am not sure that this is a Bleak error? 
A small program showing how you use bleak when this happens will suffice.

GreatScott, 2020-08-03:
Hi @hbldh -- here is a minimal example where I verified this is still occurring on 0.7.1: 

https://github.com/neosensory/neosensory-sdk-for-python/blob/master/examples/buzz_a_buzz.py (I am aware the address format in the example is currently for Windows-- am seeing the issue when I explicitly give it a known correct MacOS-formatted address. The scanner in the example will pick up the device just fine, but am still getting a hang in the same spot.

I will reach out to you offline about sending over a device if interested?

GreatScott, 2020-08-06:
Apologies: I hadn't pulled the correct branch, didn't release this wasn't in an official release yet. I can verify that the issue is now resolved for me with version 0.7.2a4 (installed with `pip install git+https://github.com/hbldh/bleak.git@develop`). Thanks a ton @hbldh !

closed 2020-08-06 by GreatScott

OneTouchOfME, 2022-09-26:
I have the same issue with hanging at line bleakClient.connect(): I have asked a question at stackoverflow. Can someone help me? [Hanging bleak](https://stackoverflow.com/questions/73849910/hangling-of-bleakclient-connect-in-continous-connect-and-disconnect-with-ble-dev)

MateoSegura, 2022-09-30:
Same issue still happening under Windows 11. Using the default UART Service example.

dlech, 2022-09-30:
Since this issue has been closed, can you please start a new issue with a reproducible test case and logs (BLEAK_LOGGING and wireshark bluetooth capture)?

commit 7dd4d3a6 (2020-07-20): Fix for connected_peripheral_peripheral problem

Closes #213.
Fixes #116.