hbldh/bleak #118: Not able to access BluetoothLEAdvertisement data

vk117, 2019-10-14:
Hi,

I was going through the documentation and I got kind of confused where it said that details attribute is a Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisement object.

But when I try to access the advertisement data, I get System.__ComObject instead. Does it mean that I cannot access the raw BLE Advertisement data in bytes?

Is there any way to access the advertisement data directly?

Any help is appreciated.

Thanks

hbldh, 2019-11-01:
Sorry for the late reply, but here is something that might help if you are still interested.

`pythonnet` does not really recognise the UWP objects in the `Windows` namespace, so you have to type them by yourself: 

```
BluetoothLEAdvertisementType(<your System.__ComObject object>)
```

I get `System.__ComObject` back in e.g. [`read_gatt_char`][https://github.com/hbldh/bleak/blob/master/bleak/backends/dotnet/client.py#L308] and I retype the `GattCommunicationStatus.Value` object to a `IBuffer`

Try it out and please close this issue if it works.

closed 2020-01-05 by hbldh

aaronfultonnz, 2020-08-24:
I'm interested in reading the raw advertisement data too. I don't quite understand what you mean by retyping into an IBuffer. Any chance of an example?

I've got this far:
`import asyncio
from bleak import BleakScanner

# See https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisement?view=winrt-19041
def detection_callback(*args):
    device = args[1]
    print(device.AdvertisementType)  # Works
    print(device.BluetoothAddress)  # Works
    print(device.Advertisement)  # returns 'Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisement'
    print(device.Advertisement.DataSections)  # returns 'System.__ComObject'

async def run():
    scanner = BleakScanner()
    scanner.register_detection_callback(detection_callback)
    await scanner.start()
    await asyncio.sleep(5.0)
    await scanner.stop()

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
`

hbldh, 2020-09-02:
@aaronfultonnz The `device.Advertisement.DataSections` should be a list of [BluetoothLEAdvertisementDataSection](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementdatasection?view=winrt-19041)

Try something like this:

```python
from System import Array, Byte
from Windows.Devices import Enumeration
from Windows.Devices.Bluetooth.Advertisement import BluetoothLEAdvertisementDataSection
from Windows.Storage.Streams import DataReader, IBuffer

def detection_callback(*args):
    device = args[1]
    print('---')
    print(device.AdvertisementType)  # Works
    print(device.BluetoothAddress)  # Works
    print(device.Advertisement)  # returns 'Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisement'
    # device.Advertisement.DataSections looks like a System.__ComObject but is actually an
    # IList[BluetoothLEAdvertisementDataSection] so we can loop over it.
    for data_section in device.Advertisement.DataSections:  # data_section is now a 'Windows.Devices.Bluetooth.Advertisement.BluetoothLEAdvertisementDataSection'
        buffer = IBuffer(data_section.Data)  # data_section.Data is also a System.__ComObject, but we know that it should be an IBuffer. Casting to that type here. 
        b = Array.CreateInstance(Byte, buffer.Length)  # Create a C# array to read the data from the buffer into.
        reader = DataReader.FromBuffer(buffer)  # Create a reader for the buffer
        reader.ReadBytes(b)  # Read
        reader.DetachBuffer()  # Release the buffer. Might not be needed.
        reader.Dispose()  # Dispose of the reader object.
        print(bytes(b))
```

It helps to learn some C# if you want to delve deeper. `pythonnet`, which handles the Python/CLR conversions cannot manage all types, so one has to do some manual type casting and handling in some cases. This is one of them.

aaronfultonnz, 2020-09-05:
Thanks @hbldh that works perfect!