hbldh/bleak #120: Error in write_gatt_char() error= an integer is required (got type NoneType)

circuitdevelopers-in, 2019-10-25:
* bleak version: 0.5.1
* Python version: 3.6.8
* Operating System: Ubuntu 18 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

I am try to write data using write_gatt_char() function and the error is coming
an integer is required (got type NoneType)

### What I Did
My Code->

#address = "00:A0:50:56:89:A6"
from os import *
import asyncio
from bleak import discover
from bleak import BleakClient
address = "00:A0:50:56:89:A6"
MODEL_NBR_UUID = "00005343-0000-1000-8000-00805f9b34fb"

async def run(address, loop):
    async with BleakClient(address, loop=loop) as client:
        status = await client.is_connected()
        print(status)
        if(status==True):
            write_value = bytearray([0x08])
            print(write_value)

            value=await client.write_gatt_char(MODEL_NBR_UUID, write_value)
            print(value)
            for service in client.services:
                print("[Service] {0}: {1}".format(service.uuid, service.description))
                for char in service.characteristics:
                    print("properties={}, uuid={}".format(char.properties,char.uuid))
        else:
            print("not connected")

loop = asyncio.get_event_loop()
loop.run_until_complete(run(address, loop))

OUTPUT->
```
True
bytearray(b'\x08')
Traceback (most recent call last):
  File "ble.py", line 31, in <module>
    loop.run_until_complete(run(address, loop))
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "ble.py", line 21, in run
    value=await client.write_gatt_char(MODEL_NBR_UUID, write_value, response=False)
  File "/home/puneet/Downloads/stream-video-browser/myenv/lib/python3.6/site-packages/bleak/backends/bluezdbus/client.py", line 427, in write_gatt_char
    os.write(fd, data)
TypeError: an integer is required (got type NoneType)

```

hbldh, 2019-11-01:
This is most probably an error due to bug described in and circumvented by #122. Look at the comment [here](https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/client.py#L414-L417).

In version 0.5.2 this will be remedied. In the meantime, run `pip install txdbus==1.1.2` and try again. Please close this issue if it works; good luck!

dlech, 2019-11-01:
> `pip install txdbus==1.1.2`

I think it is just version 1.1.1

hbldh, 2019-11-01:
@dlech My mistake; short memory apparently. Thanks!

hbldh, 2020-06-02:
I think this has been solved. Closing this issue.

closed 2020-06-02 by hbldh