hbldh/bleak #121: Reconnection logic

obalenko, 2019-10-29:
* bleak version: 0.5.1
* Python version: 3.6.8
* Operating System: Ubuntu
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.51

### Description
I'm trying to implement reconnection logic in my python script. The idea is the following: 
1) Connecting to specific device 
2) Reading / writing / listening for notification from this device
3) In case the device suddenly disconnects, start from beginning (waiting for connection and (2))

### What I Did
I tried to start a new loop if I receive disconnection callback but it doesn't seems to work for me

How I can manage the reconnect process?

Thanks in advance!
```
async def run(loop, debug=False):
    '''
    Async entry point
    :param loop: async loop
    :param debug: enable/disable debugger
    :return: None
    '''
    global CONNECTION_STATE
    if debug:

        # loop.set_debug(True)
        l = logging.getLogger("asyncio")
        l.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        l.addHandler(h)
        logger.addHandler(h)

    address = None

    try:
        if MAC_ADDRESS_TYPE == "auto":
            device = None
            while device is None:
                devices = await discover()
                device = next((d for d in devices if CHARACTERISTIC_UUID_PA_SERVICE in d.details["props"].get("UUIDs")), None)
                try:
                    log_message("Discover", "name: ({}) address: ({}) was found".format(device.name, device.address), syslog.LOG_INFO)
                except AttributeError:
                    log_message("Discover", "Device not found",
                                syslog.LOG_INFO)
                    continue
                address = device.address

        if MAC_ADDRESS_TYPE == "list":
            device = None
            while device is None:
                devices = await discover()
                device = next((d for d in devices if get_mac_from_file(d.address)), None)
                try:
                    log_message("Discover", "address: ({}) was found in whitelist".format(device.address),
                            syslog.LOG_INFO)
                except AttributeError:
                    log_message("Discover", "Device not found",
                                syslog.LOG_INFO)
                    continue
                address = device.address

        CONNECTION_STATE = True

        async with BleakClient(address, loop=loop) as client:

            def disconnected_callback(client: BleakClient):
                '''
                Disconnect callback logger
                :param client: BleakClient
                :return: None
                '''
                global CONNECTION_STATE
                log_message("disconnect()", "Disconnected", syslog.LOG_INFO)
                led_state("off")
                CONNECTION_STATE = False

            client.set_disconnected_callback(disconnected_callback)
            x = await client.is_connected()
            await client.start_notify(CHARACTERISTIC_UUID_TX_RX, notification_handler)
            auth_status = False

            try:
                auth_status = await asyncio.wait_for(start_authentication(client, pkt_1_challenge, pkt_2_challenge, challenge_message), timeout=15)
            except (asyncio.TimeoutError):
                log_message("Authentication", "authentication timeout. disconnecting...", syslog.LOG_INFO)

            if auth_status:
                led_state("on")

                await client.start_notify(CHARACTERISTIC_UUID_PA_MUTE, notification_handler)

                while True:
                    if not CONNECTION_STATE:
                        await client.disconnect()
                        break
            else:
                led_state("off")

    except Exception as e:
        log_message("Error (Sierra)", str(e), syslog.LOG_ERR)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run(loop, False))

```

hbldh, 2019-11-01:
First, do not use a global state for connectedness. Either ask the BleakClient object if it is connected before you try to use it or try to handle it like in the [disconnect_callback.py example] (https://github.com/hbldh/bleak/blob/master/examples/disconnect_callback.py).

My suggestion is to throw the instance away and create a new one when you then want to reconnect.

closed 2019-11-01 by hbldh

JBrakefield, 2021-05-25:
I am similarly stuck on reconnection logic, and the example disconnect_callback.py does not seem to solve that problem (I don't understand what it is actually doing).

Did you ever figure this out? I'm going in circles because the disconnection callback is not an asyn function, so I can't get it to call the connection function again. I've tried ending the loop and starting a new one, but that keeps failing as well.

sembree, 2022-09-23:
I was able to modify the script above to reconnect after disconnect.  In the disconnect callback, cancel all asyncio tasks as follows:  

```
def handle_disconnect(_: BleakClient):
    # cancelling all tasks ends the asyncio event loop
    for task in asyncio.all_tasks():
            task.cancel()
```

This will end the asyncio event loop.  Then in the main function, restart the event loop.

In the "while True" loop, use "await asnycio.sleep(1)" to keep the event loop alive while the connection is active.