hbldh/bleak #125: discover and connect never complete from non-MainThread

derekgorcz, 2019-11-06:
* bleak version:
0.5.0
* Python version:
3.6.8
* Operating System:
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic

* BlueZ version (`bluetoothctl -v`) in case of Linux: 
bluetoothctl: 5.50

### Description
discover and connect never complete from non-mainthread

It appears that this may be failing to connect to the system dbus: https://github.com/hbldh/bleak/blob/master/bleak/backends/bluezdbus/discovery.py#L128

### What I Did

```
import asyncio
import threading
from bleak import discover

async def run(loop):
    print('1'*80)
    devices = await discover(loop=loop)  # discover never completes
    print('2'*80)
    for d in devices:
        print(d)

def threaded_func():
    # loop = asyncio.get_event_loop()  # this will fail with "RuntimeError: There is no current event loop in thread 'Thread-1'."
    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)
    loop.run_until_complete(run(loop))

thread = threading.Thread(target=threaded_func)
thread.start()
thread.join()
```

hbldh, 2019-11-07:
Similar to #93. You should not create a new event loop using the BlueZ. Use the `asyncio.get_event_loop` and skip the `threading` module. Are you sure you really need it? If you are using the `asyncio` module, then you are better off using it for all you distributed tasks.

Use something like this instead:

```python
import asyncio
from bleak import discover

async def run(loop):
    print('1'*80)
    devices = await discover(loop=loop) 
    print('2'*80)
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run(loop))
```

derekgorcz, 2019-11-07:
I am not able to execute from the main thread. As a workaround I am currently having my thread spin up a new process.

closed 2020-06-02 by hbldh