hbldh/bleak #127: _uuid (str or UUID)

dlech, 2019-11-11:
* bleak version: develop
* Python version: N/A
* Operating System: N/A

### Description

There are several places in the documentation where there is a `_uuid` parameter. The docs say:

    _uuid (str or UUID): The uuid of the characteristics to write to.

However, the type hints say: `_uuid: str`.

Looking at the actual code, in some cases, `uuid.UUID` is handled, but in others it is assumes that the parameter is a string, so when a pass a `uuid.UUID`, things crash.

Which is correct, the doc string or the type hints?

hbldh, 2019-12-27:
I had planned for both `uuid.UUID` and `str` to be able to be handled, and all places where the are not both useable should be changed. Type hints should also be modified to correspond to the duality of the input.

5frank, 2019-12-30:
Just adding 2 cents to the discussion. 
Support for both `uuid.UUID` and `str` seems to add a lot of complexity and will sometimes get confusing. Example when using `start_notify()` with `uuid.UUID` but the callback (aka response_handler) is called with the same uuid as `str`. Would be nice if it was `uuid.UUID` as it provides better sanity check and comparison is not case sensitive.

hbldh, 2020-01-05:
The complexity of handling both is not that great. The problem is that I built everything around strings in the beginning, since BlueZ handles only strings and I found it more convenient to convert Windows Guid objects to strings than to uuid objects. 

I am making an effort to make both possible to use now and then try to do the swap behind the scenes later on.

5frank, 2020-01-12:
just an idea:
```python
STR_A='00000000-1234-abcd-1234-00000000000a'
STR_B='00000000-1234-abcd-1234-00000000000b'
    
class BleakUUID(UUID):
    ''' Same as UUID but compares to string '''
    def __cmp__(self, other):
        if not isinstance(other, UUID):
            try:
                other = UUID(other)
            except ValueError:
                return False
        return self.int - other.int

    def __eq__(self, other):
        return self.__cmp__(other) == 0
    def __ne__(self, other):
        return self.__cmp__(other) != 0
    def __gt__(self, other):
        return self.__cmp__(other) > 0
    def __lt__(self, other):
        return self.__cmp__(other) < 0
    def __ge__(self, other):
        return self.__cmp__(other) >= 0
    def __le__(self, other):
        return self.__cmp__(other) <= 0

def test():
    a = BleakUUID(STR_A)
    assert(isinstance(a, UUID) == True)
    assert(a == STR_A)
    assert(a != STR_B)
    assert(a < STR_B)
```

hbldh, 2020-01-18:
It can be done like this, yes, but the problem is not handling the comparisons. Thanks for the effort, but this will not add enough value to be worthwhile imo.

commit 178346f5 (2019-12-27): Improved type hints and handling of str/uuid

Fixes #127 for Windows and BlueZ backends.
commit 13014a68 (2020-03-09): Improved type hints and handling of str/uuid

Fixes #127 for Windows and BlueZ backends.
PR #164 (2020-03-09): Release/v0.6.0
* New Scanner object to allow for async device scanning.
* Updated ``txdbus`` requirement to version 1.1.1 (Merged #122)
* Implemented ``write_gatt_descriptor`` for Bluez backend.
* Large change in Bluez backend handling of Twisted reactors. Fixes #143
* Modified ``set_disconnect_callback`` to actually call the callback as a callback. Fixes #108.
* Added another required parameter to disconnect callbacks.
* Added Discovery filter option in BlueZ backend (Merged #124)
* Merge #138: comments about Bluez version check.
* Improved scanning data for macOS backend. Merge #126.
* Merge [... cut]
closed 2020-03-09 by hbldh